ROBOTIS-GIT/turtlebot3 #789: How do I make changes to the OpenCR code?

royito55, 2021-10-05:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

9. Please describe the issue in detail.

I am trying to get rid of the low voltage warning buzzer. I realized that the odometry doesn't work after this. 

I followed these instructions to find where I need to deactivate the alarm:

https://answers.ros.org/question/296145/turtlebot3-low-battery-beeping/?comment=387904#post-id-387904

But I don't know the actual steps to reupload the opencr packages? The manual simply says

How to modify ROS library code
You can modify ROS library in below paths.
Your/Arduino/Board/Package/Directory/OpenCR/hardware/OpenCR/Version/libraries/turtlebot3
Your/Arduino/Board/Package/Directory/OpenCR/hardware/OpenCR/Version/libraries/turtlebot3_ros_lib
If you want to modify it in the sketch folder, move the above two folders to the “skechbook/libraries” folder.

And I've found that path, but I can't open the .cpp in the Arduino IDE.

How do I flash these changes to the OpenCR?

ashekim, 2021-10-05:
Hi @royito55. Thank you for your questions.
I think you can find your library folder path in Arduino > Preferences.

[image]

In addition, if you refer to the [link](https://learn.adafruit.com/adafruit-all-about-arduino-libraries-install-use/how-to-install-a-library), detailed explanations from library addition to upload are provided.

Thank you

royito55, 2021-10-05:
Hi, thank you for answering. 

I don't need to know where my library folder path is, I've already found it. 

My question is, once I know where to change the code, how do I reupload these .cpp files into the OpenCR?

royito55, 2021-10-05:
To clarify. I have installed the OpenCR in my Arduino IDE. Now, I want to modify that library to deactivate the low voltage alarm, and upload the changes to the physical OpenCR. How can I do this?

royito55, 2021-10-06:
Thanks for the info. However, I'm still confused.

The link you've shared is showing me how to install a library in the Arduino IDE in my Linux system, and nowhere does it say how to upload to the Arduino. I'm guessing any sketch that is uploaded using a library means the library gets uploaded?

Your GIF is really helpful though, although you are including a library as a .zip file, where the link specifies it should be extracted.

I did what you are doing in your gif, opened the turtlebot3_core example and then include the modified libraries through the add .zip libraries, in the correct location. Then I compiled and uploaded to the board. 

Is that everything I need to do? 
How can I check if the changes have been implemented in the OpenCR?

sunghowoo, 2024-11-13:
Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. This issue will be closed due to prolonged inactivity.  if the issue persists, feel free to reopen it.

closed 2024-11-13 by sunghowoo