ROBOTIS-GIT/turtlebot3 #797: Incorrect USB Port Setting for OpenCR

srushtibobade, 2021-11-05:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

   
   
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

   
    - [ ] ROS 2 Dashing Diademata
 

3. Which SBC(Single Board Computer) is working on TurtleBot3?

  
    - [ ] Raspberry Pi 4
  

4. Which OS you installed on SBC?

 
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
   

5. Which OS you installed on Remote PC?

    
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
   

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.
sudo cp `ros2 pkg prefix turtlebot3_bringup`/share/turtlebot3_bringup/script/99-turtlebot3-cdc.rules /etc/udev/rules.d/

8. Copy and Paste the error messages on terminal.

   Command not found 

9. Please describe the issue in detail.

 Trying to configure TB3 for ROS2 dashing and USB port setting for Open CR is not working.
Can this be a correct command ./99-turtlebot3-cdc.rules /etc/udev/rules.d/?

ashekim, 2021-11-09:
Hello. Thanks for the question.
Please click the dashing tab in the [link](https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup-3) below and follow them in order

Thank you.

srushtibobade, 2021-11-23:
Hi @ashekim
Sorry for the late reply. But the first command of step 6 of SBS set up which is : sudo cp `ros2 pkg prefix turtlebot3_bringup`/share/turtlebot3_bringup/script/99-turtlebot3-cdc.rules /etc/udev/rules.d/ is incorrect.

Can you please help me by providing the right command?

ROBOTIS-Will, 2021-11-24:
Hi @srushtibobade 

Sorry about the incorrect command for Dashing.
Please use the command below as the directory has been reorganized since Foxy.
```
$ sudo cp `ros2 pkg prefix turtlebot3_bringup`/share/turtlebot3_bringup/99-turtlebot3-cdc.rules /etc/udev/rules.d/
```
Thank you.

srushtibobade, 2021-11-26:
Hi @ROBOTIS-Will 

Unfortunately, the command you provided did not work as well.
I believe this is the correct command (I used this one by looking at the directory structure):

$ sudo cp ./99-turtlebot3-cdc.rules /etc/udev/rules.d/

ROBOTIS-Will, 2021-11-29:
Hi @srushtibobade 

If you have specified the location of the file, it will work too.
As long as the correct file is copied into the `/etc/udev/rules.d/` directory, it should be okay.
Thanks!

srushtibobade, 2021-12-01:
Hi @ROBOTIS-Will 

Thanks!!

closed 2021-12-01 by srushtibobade