ROBOTIS-GIT/turtlebot3 #800: Using global_palnner for A star algorithm makes the planned path stick to walls and collide.

keshri-prasanjeet, 2021-11-22:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - Burger

2. Which ROS is working with TurtleBot3?

    - ROS 1 Melodic Morenia

3. Which SBC(Single Board Computer) is working on TurtleBot3?

   - just simulating in RVIZ and Gazebo

4. Which OS you installed on SBC?

    - Ubuntu preinstalled server (18.04/20.04)

5. Which OS you installed on Remote PC?

   - Just simulating

6. Specify the software and firmware version(Can be found from Bringup messages)

```
    PARAMETERS
     * /rosdistro: melodic
     * /rosversion: 1.14.12
     * /turtlebot3_core/baud: 115200
     * /turtlebot3_core/port: /dev/ttyACM0
     * /turtlebot3_core/tf_prefix:
     * /turtlebot3_lds/frame_id: base_scan
     * /turtlebot3_lds/port: /dev/ttyUSB0
    
    NODES
      /
        turtlebot3_core (rosserial_python/serial_node.py)
        turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
        turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)
    
    ROS_MASTER_URI=http://localhost:11311
```

7. Specify the commands or instructions to reproduce the issue.

Trying to use the A star algorithm from GlobalPlanner for global planing and not the dijkstra or NavFn.
These are my parameters for implementing GlobalPlanner
```
          GlobalPlanner:                                 
          old_navfn_behavior: false                    
          use_quadratic: true                          
          use_dijkstra: false                           
          use_grid_path: true                         
        
          allow_unknown: true                          
        
          planner_window_x: 0.0                        
          planner_window_y: 0.0                        
          default_tolerance: 0.0                       
        
          publish_scale: 100                           
          planner_costmap_publish_frequency: 0.0       
        
          lethal_cost: 253                             
          neutral_cost: 50                             
          cost_factor: 3.0                             
          publish_potential: true
```

I set base_global_planner as `global_planner/GlobalPlanner`. to use the a star algorithm is set `use_dijkstra` to `false` and `use_grid_path` to `true`

8. Copy and Paste the error messages on terminal.

    
[image]

9. Please describe the issue in detail.
Upon trying to use the a star algorithm the turtlebot sticks to walls. and ends up colliding a lot and getting stuck in the corners. 
By simply reverting back to NavFn and not using GlobalPlanner or turning use_dijkstra to true again the issue is no more. How can I implement a proper A* algorithm.

ashekim, 2021-12-06:
Hello. @keshri-prasanjeet 
You can get more answers to questions about navigation algorithms in the ros forum. 
Please contact us in the [ros forum](https://answers.ros.org/questions/).

thank you.

sunghowoo, 2024-11-08:
This issue will be closed due to prolonged inactivity.

closed 2024-11-08 by sunghowoo