ROBOTIS-GIT/turtlebot3 #801: turtlebot3 using only DYNAMIXEL U2D2?!?!?

3mrrrx, 2021-11-22:
would it be possible to control turtlebot3 using only DYNAMIXEL U2D2?

I am build my turtlebot3 for old parts, and using the U2D2 should be very doable in theory, but I can not find an implantation for this. In theory using U2D2 would be even better, as the OpenCR has to split its processor time between publishing, sensor reading and running the motors.

The OpenCR code apparently also does some minor controls for the messages coming from the PI. so these have to be also handled within the controller node on the pi... please correct me if i am wrong here.

looking for an implantation in ROS2-Foxy or Galactic.

thankful on any help in this regard

ashekim, 2021-12-06:
HI. @3mrrrx. Thanks for your interest in turtlebot3.
OpenCR has the ability to control DYNAMIXEL. And OpenCR has various sensors including IMU. 
The IMU sensor is used when calculating the odom of TB3. 

So unfortunately, it is difficult to control TurtleBot3 with only U2D2.
Sorry it wasn't the answer you were looking for.

thank you.

3mrrrx, 2021-12-06:
regarding the IMU and maybe the other sensor, these can be including in ROS separately and would published to the NAV or any other ROS node.

i also looked at the OpenCR code... it should be possible to implement the hole thing in python.. its quite a lot of work, but would also open a lot of possibilities.

ashekim, 2021-12-07:
@3mrrrx 

Hi again.
If you check the link below, you can check the specifications of OpenCR.
- [OpenCR Spec](https://emanual.robotis.com/docs/en/parts/controller/opencr10/#specifications)

The OpeCR board contains various sensors including the IMU sensor.
If you only use U2D2, it means that you cannot get the IMU sensor value even if you modify the code.
Therefore, if you use U2D2 instead of OPENCR, you need a additional imu sensor.

sunghowoo, 2024-11-08:
This issue will be closed due to prolonged inactivity.

closed 2024-11-08 by sunghowoo