ROBOTIS-GIT/turtlebot3 #802: I want to make another larger robot

ROBOTIS-Will, 2021-11-23:
Hi 

My turtlebot3 waffle is working well and I appreciate the work you guys have done to make this happen.

Now I want to make another larger robot and  I have a few questions for you. This robot will have large wheels say 6 inch and be able to drive all around my house without having problems with the carpet like the waffle does

Can you recommend a motor controller and motor that can take 6 inch wheels that have good encoders built in that can be used.

The laptop I use for the TurtleBot3 has ubuntu 16.04 loaded on it with ros kinetic 
The raspberry pi 3B+ is set up with ubuntu 16.04 and kinetic as well

Regarding the releases of ubuntu and ros as time has moved on and I am making a larger robot should I use later releases of ubuntu and ros or stay with the 16.04 and kinetic versions as I know they work?

Also do the versions on the remote laptop and raspberry pi 3B+ (SBC) of ubuntu and ros need to be the same  or can they be different ?
this would be like, I keep ubuntu 16.04 on the laptop and update the raspberry pi to sat 18.04 or 21.1. Or must they be the same on both

regards
Max

> On 16 Nov 2020, at 20:37, ROBOTIS-Ashe ***@***.***> wrote:
> 
> 
> Hi. again. :)
> 
> It seems like an error that occurs when running roslaunch turtlebot3_bringup turtlebot3_remote.launch on the PC.
> 
> Please refer to the link below
> turtlebot3_remote.launch <https://github.com/ROBOTIS-GIT/turtlebot3/blob/ba10294ccb02891cd6af456bd4662e77813e3245/turtlebot3_bringup/launch/turtlebot3_remote.launch#L5>
> Have you ever run roslaunch turtlebot3_bringup turtlebot3_remote.launch?
> 
> If so, you are supposed to type TB3_MODEL.
> $ export TURTLEBOT3_MODEL=waffle_pi
> 
> Please Give it a try and let me know the result.
> Thank you.
> 
> —
> You are receiving this because you were mentioned.
> Reply to this email directly, view it on GitHub <https://github.com/ROBOTIS-GIT/turtlebot3/issues/674#issuecomment-728648066>, or unsubscribe <https://github.com/notifications/unsubscribe-auth/AD5FYGDCIY2YHTPTMQNZLILSQHOVVANCNFSM4TVEHLMA>.
> 

Warmest Regards
 
Max Birley

Check this out
https://youtu.be/oGikNKcbI0

Please respond to all emails from ***@***.*** and reply to  ***@***.*** as ***@***.*** will not be operable by late November

Cell +1 830 998 9914

Our Web Site is you no mow
http://www.younomow.com/

_Originally posted by @Mdbirley in https://github.com/ROBOTIS-GIT/turtlebot3/issues/674#issuecomment-975962989_

ROBOTIS-Will, 2021-11-25:
You might be interested in TurtleBot3 Carrier below.
https://emanual.robotis.com/docs/en/platform/turtlebot3/locomotion/#turtlebot3-friends-carrier

A custom wheel design is available in the 3D modeling so that you can use 3D print the part
The ROS version on Remote PC and TurtleBot SBC should be the same as different distributions don't guarantee compatibility.

Thanks!

Mdbirley, 2022-01-27:
Hi
I have a turtlebot3 and I am trying to get the slam in rviz and the raspberry pi image at the same time
I do the following

Laptop run rescore
Robot run roslaunch turtlebot3_bringup turtlebot3.rpicamera.launch
laptop run roslaunch turtlebot3_slam turtlebot3_slam.launch

rviz opens I get an image from the camera but no lidar data

Can you help by providing the exact commands to also get the lidar data

regards
Max

> 
> This email originated from outside of 5 Hertford Street. Do not click on any links or open attachments unless you recognise the sender and know the content is safe.
> 
> 
> 
> 
> 
> 
> You might be interested in TurtleBot3 Carrier below.
> https://emanual.robotis.com/docs/en/platform/turtlebot3/locomotion/#turtlebot3-friends-carrier <https://linkscan.io/scan/ux/aHR0cHM6Ly9lbWFudWFsLnJvYm90aXMuY29tL2RvY3MvZW4vcGxhdGZvcm0vdHVydGxlYm90My9sb2NvbW90aW9uLyN0dXJ0bGVib3QzLWZyaWVuZHMtY2Fycmllcg==/C1ED417BB6BE5C1A8236D472AF5634ED6D58213946E3A29B25F65165387CB2E5?c=5&i=1&docs=1>
> A custom wheel design is available in the 3D modeling so that you can use 3D print the part
> The ROS version on Remote PC and TurtleBot SBC should be the same as different distributions don't guarantee compatibility.
> 
> Thanks!

Warmest Regards
 
Max Birley

Check this out
https://youtu.be/oGikNKcbI0

Please respond to all emails from ***@***.*** and reply to  ***@***.*** as ***@***.*** will not be operable by late November

Cell +1 830 998 9914

Our Web Site is you no mow
http://www.younomow.com/

ROBOTIS-Will, 2022-01-27:
Hi @Mdbirley 

> Robot run roslaunch turtlebot3_bringup turtlebot3.rpicamera.launch

The turtlebot3_rpicamera.launch only activates the Raspi Camera.
In order to run the SLAM, you'll need to activate the LDS sensor and other necessary functions from TurtleBot3 with below command.
```
$ roslaunch turtlebot3_bringup turtlebot3_robot.launch
```

If you wish to run both camera and basic features, you can merge the content in the `turtlebot3_rpicamera.launch` into the `turtlebot3_robot.launch` as below.
```
<launch>
  <arg name="multi_robot_name" default=""/>
  <arg name="set_lidar_frame_id" default="base_scan"/>

  <include file="$(find turtlebot3_bringup)/launch/turtlebot3_core.launch">
    <arg name="multi_robot_name" value="$(arg multi_robot_name)"/>
  </include>
  <include file="$(find turtlebot3_bringup)/launch/turtlebot3_lidar.launch">
    <arg name="set_frame_id" value="$(arg set_lidar_frame_id)"/>
  </include>

  <node pkg="turtlebot3_bringup" type="turtlebot3_diagnostics" name="turtlebot3_diagnostics" output="screen"/>
  
  <node pkg="raspicam_node" type="raspicam_node" name="raspicam_node" output="screen">
    <param name="camera_info_url" value="package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml"/>
    <param name="width" value="640"/>
    <param name="height" value="480"/>
    <param name="framerate" value="15"/>
    <param name="camera_frame_id" value="camera"/>
  </node>
</launch>
```

Thank you.

Mdbirley, 2022-01-27:
very fast response…that is appreciated

I will try this in the morning

thanks

> 
> This email originated from outside of 5 Hertford Street. Do not click on any links or open attachments unless you recognise the sender and know the content is safe.
> 
> 
> 
> 
> 
> 
> Hi @Mdbirley <https://linkscan.io/scan/ux/aHR0cHM6Ly9naXRodWIuY29tL01kYmlybGV5/3754A46C6CF10C6E6FD7479D2FF5B1FFF47B39562A81665A921475815CE26C6A?c=5&i=1&docs=1>
> Robot run roslaunch turtlebot3_bringup turtlebot3.rpicamera.launch
> 
> The turtlebot3_rpicamera.launch only activates the Raspi Camera.
> In order to run the SLAM, you'll need to activate the LDS sensor and other necessary functions from TurtleBot3 with below command.
> 
> $ roslaunch turtlebot3_bringup turtlebot3_robot.launch
> If you wish to run both camera and basic features, you can merge the content in the turtlebot3_rpicamera.launch into the turtlebot3_robot.launch as below.
> 
> <launch>
>   <arg name="multi_robot_name" default=""/>
>   <arg name="set_lidar_frame_id" default="base_scan"/>
> 
>   <include file="$(find turtlebot3_bringup)/launch/turtlebot3_core.launch">
>     <arg name="multi_robot_name" value="$(arg multi_robot_name)"/>
>   </include>
>   <include file="$(find turtlebot3_bringup)/launch/turtlebot3_lidar.launch">
>     <arg name="set_frame_id" value="$(arg set_lidar_frame_id)"/>
>   </include>
> 
>   <node pkg="turtlebot3_bringup" type="turtlebot3_diagnostics" name="turtlebot3_diagnostics" output="screen"/>
>   
>   <node pkg="raspicam_node" type="raspicam_node" name="raspicam_node" output="screen">
>     <param name="camera_info_url" value="package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml"/>
>     <param name="width" value="640"/>
>     <param name="height" value="480"/>
>     <param name="framerate" value="15"/>
>     <param name="camera_frame_id" value="camera"/>
>   </node>
> </launch>
> Thank you.

Warmest Regards
 
Max Birley

Check this out

https://youtu.be/oGikNKcbIr

Cell +1 830 998 9914

http://www.younomow.com/

Mdbirley, 2022-01-27:
Hi again

I copied the code that you posted below into  turtlebot3_robot.launch as below.
I then ran roslaunch turtlebot3_bringup turtlebot3_robot.launch

I then ran the slam launch file

I got the lidar data in rviz but not camera image

Do I need to compile the code if so can you assist with the actual compile commands and directories.

max
> <launch>
>   <arg name="multi_robot_name" default=""/>
>   <arg name="set_lidar_frame_id" default="base_scan"/>
>
>   <include file="$(find turtlebot3_bringup)/launch/turtlebot3_core.launch">
>     <arg [...]

ROBOTIS-Will, 2022-01-27:
@Mdbirley 
Could you confirm the list of topics that TurtleBot3 publishes from your laptop?
```
$ rostopic list
```

Mdbirley, 2022-01-30:
I copied your launch code in to the bringup launch file

Ran rescore on the laptop
ssh into the tb3 and ran the bring up

opened another terminal on the tb3 and ran rostopic list and got the following

then ran rostopic on the laptop and got the same list

then ran slam on the laptop

then ran rostopic list and got this 

sill not sign of the raspberry pi camera. It 
would be useful so I can steer the TB3 without needing to follow it

looking forward to the solution

regards

Max

this is the new 

<launch>
  <arg name="multi_robot_name" default=""/>
  <arg name="set_lidar_frame_id" default="base_scan"/>

  <include file="$(find turtlebot3_bringup)/launch/turtlebot3_core.launch">
    <arg name="multi_robot_name" value="$(arg multi_robot_name)"/>
  </include>
  <include file="$(find turtlebot3_bringup)/launch/turtlebot3_lidar.launch">
    <arg name="set_frame_id" value="$(arg set_lidar_frame_id)"/>
  </include>

  <node pkg="turtlebot3_bringup" type="turtlebot3_diagnostics" name="turtlebot3_diagnostics" output="screen"/>
  
  <node pkg="raspicam_node" type="raspicam_node" name="raspicam_node" output="screen">
    <param name="camera_info_url" value="package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml"/>
    <param name="width" value="640"/>
    <param name="height" value="480"/>
    <param name="framerate" value="15"/>
    <param name="camera_frame_id" value="camera"/>
  </node>
</launch>

>> Hi @Mdbirley <https://linkscan.io/scan/ux/aHR0cHM6Ly9naXRodWIuY29tL01kYmlybGV5/3754A46C6CF10C6E6FD7479D2FF5B1FFF47B39562A81665A921475815CE26C6A?c=5&i=1&docs=1>
>> Robot run roslaunch turtlebot3_bringup turtlebot3.rpicamera.launch
>> 
>> The turtlebot3_rpicamera.launch only activates the Raspi Camera.
>> In order to run the SLAM, you'll need to activate the LDS sensor and other necessary functions from TurtleBot3 with below command.
>> 
>> $ roslaunch turtlebot3_bringup turtlebot3_robot.launch
>> If you wish to run both camera and basic features, you can merge the content in the turtlebot3_rpicamera.launch into the turtlebot3_robot.launch as below.
>> 
>> <launch>
>>   <arg name="multi_robot_name" default=""/>
>>   <arg name="set_lidar_frame_id" default="base_scan"/>
>> 
>>   <include file="$(find turtlebot3_bringup)/launch/turtlebot3_core.launch">
>>     <arg name="multi_robot_name" value="$(arg multi_robot_name)"/>
>>   </include>
>>   <include file="$(find turtlebot3_bringup)/launch/turtlebot3_lidar.launch">
>>     <arg name="set_frame_id" value="$(arg set_lidar_frame_id)"/>
>>   </include>
>> 
>>   <node pkg="turtlebot3_bringup" type="turtlebot3_diagnostics" name="turtlebot3_diagnostics" output="screen"/>
>>   
>>   <node pkg="raspicam_node" type="raspicam_node" name="raspicam_node" output="screen">
>>     <param name="camera_info_url" value="package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml <package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml>"/>
>>     <param name="width" value="640"/>
>>     <param name="height" value="480"/>
>>     <param name="framerate" value="15"/>
>>     <param name="camera_frame_id" value="camera"/>
>>   </node>
>> </launch>
>> Thank you.
>> 
>> 
> 
> Warmest Regards
>  
> Max Birley
> 
> Check this out
> 
> https://youtu.be/oGikNKcbIr <https://youtu.be/oGikNKcbIr>
> 
> Cell +1 830 998 9914
> 
> http://www.younomow.com/

Warmest Regards
 
Max Birley

Check this out

https://youtu.be/oGikNKcbIr

Cell +1 830 998 9914

http://www.younomow.com/

Mdbirley, 2022-02-02:
do you need any other info to help you resolve the camera problem

> 
> 
> I copied your launch code in to the bringup launch file
> 
> Ran rescore on the laptop
> ssh into the tb3 and ran the bring up
> <PastedGraphic-5.png>
> opened another terminal on the tb3 and ran rostopic list and got the following
> 
> <PastedGraphic-3.png>
> 
> then ran rostopic on the laptop and got the same list
> 
> then ran slam on the laptop
>>> <PastedGraphic-4.png>
> 
> then ran rostopic list and got this 
> 
> <PastedGraphic-6.png>
> 
> 
> sill not sign of the raspberry pi camera. It 
> would be useful so I can steer the TB3 without needing to follow it
> 
> looking forward to the solution
> 
> regards
> 
> Max
> 
> this is the new 
> 
> <PastedGraphic-7.png>
> <launch>
>   <arg name="multi_robot_name" default=""/>
>   <arg name="set_lidar_frame_id" default="base_scan"/>
> 
>   <include file="$(find turtlebot3_bringup)/launch/turtlebot3_core.launch">
>     <arg name="multi_robot_name" value="$(arg multi_robot_name)"/>
>   </include>
>   <include file="$(find turtlebot3_bringup)/launch/turtlebot3_lidar.launch">
>     <arg name="set_frame_id" value="$(arg set_lidar_frame_id)"/>
>   </include>
> 
>   <node pkg="turtlebot3_bringup" type="turtlebot3_diagnostics" name="turtlebot3_diagnostics" output="screen"/>
>   
>   <node pkg="raspicam_node" type="raspicam_node" name="raspicam_node" output="screen">
>     <param name="camera_info_url" value="package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml <package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml>"/>
>     <param name="width" value="640"/>
>     <param name="height" value="480"/>
>     <param name="framerate" value="15"/>
>     <param name="camera_frame_id" value="camera"/>
>   </node>
> </launch>
> 
>>> Hi @Mdbirley <https://linkscan.io/scan/ux/aHR0cHM6Ly9naXRodWIuY29tL01kYmlybGV5/3754A46C6CF10C6E6FD7479D2FF5B1FFF47B39562A81665A921475815CE26C6A?c=5&i=1&docs=1>
>>> Robot run roslaunch turtlebot3_bringup turtlebot3.rpicamera.launch
>>> 
>>> The turtlebot3_rpicamera.launch only activates the Raspi Camera.
>>> In order to run the SLAM, you'll need to activate the LDS sensor and other necessary functions from TurtleBot3 with below command.
>>> 
>>> $ roslaunch turtlebot3_bringup turtlebot3_robot.launch
>>> If you wish to run both camera and basic features, you can merge the content in the turtlebot3_rpicamera.launch into the turtlebot3_robot.launch as below.
>>> 
>>> <launch>
>>>   <arg name="multi_robot_name" default=""/>
>>>   <arg name="set_lidar_frame_id" default="base_scan"/>
>>> 
>>>   <include file="$(find turtlebot3_bringup)/launch/turtlebot3_core.launch">
>>>     <arg name="multi_robot_name" value="$(arg multi_robot_name)"/>
>>>   </include>
>>>   <include file="$(find turtlebot3_bringup)/launch/turtlebot3_lidar.launch">
>>>     <arg name="set_frame_id" value="$(arg set_lidar_frame_id)"/>
>>>   </include>
>>> 
>>>   <node pkg="turtlebot3_bringup" type="turtlebot3_diagnostics" name="turtlebot3_diagnostics" output="screen"/>
>>>   
>>>   <node pkg="raspicam_node" type="raspicam_node" name="raspicam_node" output="screen">
>>>     <param name="camera_info_url" value="package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml <package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml>"/>
>>>     <param name="width" value="640"/>
>>>     <param name="height" value="480"/>
>>>     <param name="framerate" value="15"/>
>>>     <param name="camera_frame_id" value="camera"/>
>>>   </node>
>>> </launch>
>>> Thank you.
>>> 
>>> 
>> 
>> Warmest Regards
>>  
>> Max Birley
>> 
>> Check this out
>> 
>> https://youtu.be/oGikNKcbIr <https://youtu.be/oGikNKcbIr>
>> 
>> Cell +1 830 998 9914
>> 
>> http://www.younomow.com/ <http://www.younomow.com/>
>> 
>> 
>> 
> 
> Warmest Regards
>  
> Max Birley
> 
> Check this out
> 
> https://youtu.be/oGikNKcbIr <https://youtu.be/oGikNKcbIr>
> 
> Cell +1 830 998 9914
> 
> http://www.younomow.com/

Warmest Regards
 
Max Birley

Check this out

https://youtu.be/oGikNKcbIr

Cell +1 830 998 9914

http://www.younomow.com/

ROBOTIS-Will, 2022-02-03:
Hi @Mdbirley 
I'm sorry, but when replying via email, any attached images are abandoned so I can't see the image in your reply.
Could you copy and paste the list of the topic in your terminal output?
Thanks!

Mdbirley, 2022-02-03:
In response to your reply

> 
> 
> I copied your launch code in to the bringup launch file
> Ran roscore on the laptop
> ssh into the tb3 and ran the bring up
roslaunch turtlebot3_bringup turtlebot3_robot.launch
> 
> opened another terminal on the tb3 and ran rostopic list and got the following
> /battery state
cmd vel
cmd vel rc100
diagnostics
firmware version
imu
joint states
magnetic field
motor power
odom
reset
rosout
rosout agg
scan
sensor state
sound
tf
version info
> 
> 
> then ran rostopic on the laptop and got the similar list
> 
> then ran slam on the laptop
> 
> then ran rostopic list and got these additional topics
in addition to the above I got
initial pose
map
map metadata
map updates
motor power
move base simple/goal
tf static
turtlebot3 slam mapping/entropy
> 
> 
> 
> sill not sign of the raspberry pi camera. It 
> would be useful so I can steer the TB3 without needing to follow it
> 
> looking forward to the solution
> 
> regards
> 
> Max
> 
> this is the new 
> 
> 
> <launch>
>   <arg name="multi_robot_name" default=""/>
>   <arg name="set_lidar_frame_id" default="base_scan"/>
> 
>   <include file="$(find turtlebot3_bringup)/launch/turtlebot3_core.launch">
>     <arg name="multi_robot_name" value="$(arg multi_robot_name)"/>
>   </include>
>   <include file="$(find turtlebot3_bringup)/launch/turtlebot3_lidar.launch">
>     <arg name="set_frame_id" value="$(arg set_lidar_frame_id)"/>
>   </include>
> 
>   <node pkg="turtlebot3_bringup" type="turtlebot3_diagnostics" name="turtlebot3_diagnostics" output="screen"/>
>   
>   <node pkg="raspicam_node" type="raspicam_node" name="raspicam_node" output="screen">
>     <param name="camera_info_url" value="package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml <package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml>"/>
>     <param name="width" value="640"/>
>     <param name="height" value="480"/>
>     <param name="framerate" value="15"/>
>     <param name="camera_frame_id" value="camera"/>
>   </node>
> </launch>
> 
>>> Hi @Mdbirley <https://linkscan.io/scan/ux/aHR0cHM6Ly9naXRodWIuY29tL01kYmlybGV5/3754A46C6CF10C6E6FD7479D2FF5B1FFF47B39562A81665A921475815CE26C6A?c=5&i=1&docs=1>
>>> Robot run roslaunch turtlebot3_bringup turtlebot3.rpicamera.launch
>>> 
>>> The turtlebot3_rpicamera.launch only activates the Raspi Camera.
>>> In order to run the SLAM, you'll need to activate the LDS sensor and other necessary functions from TurtleBot3 with below command.
>>> 
>>> $ roslaunch turtlebot3_bringup turtlebot3_robot.launch
>>> If you wish to run both camera and basic features, you can merge the content in the turtlebot3_rpicamera.launch into the turtlebot3_robot.launch as below.
>>> 
>>> <launch>
>>>   <arg name="multi_robot_name" default=""/>
>>>   <arg name="set_lidar_frame_id" default="base_scan"/>
>>> 
>>>   <include file="$(find turtlebot3_bringup)/launch/turtlebot3_core.launch">
>>>     <arg name="multi_robot_name" value="$(arg multi_robot_name)"/>
>>>   </include>
>>>   <include file="$(find turtlebot3_bringup)/launch/turtlebot3_lidar.launch">
>>>     <arg name="set_frame_id" value="$(arg set_lidar_frame_id)"/>
>>>   </include>
>>> 
>>>   <node pkg="turtlebot3_bringup" type="turtlebot3_diagnostics" name="turtlebot3_diagnostics" output="screen"/>
>>>   
>>>   <node pkg="raspicam_node" type="raspicam_node" name="raspicam_node" output="screen">
>>>     <param name="camera_info_url" value="package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml <package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml>"/>
>>>     <param name="width" value="640"/>
>>>     <param name="height" value="480"/>
>>>     <param name="framerate" value="15"/>
>>>     <param name="camera_frame_id" value="camera"/>
>>>   </node>
>>> </launch>
>>> Thank you.
>>> 
>>> 
>> 
>> Warmest Regards
>>  
>> Max Birley
>> 
>> Check this out
>> 
>> https://youtu.be/oGikNKcbIr <https://youtu.be/oGikNKcbIr>
>> 
>> Cell +1 830 998 9914
>> 
>> http://www.younomow.com/ <http://www.younomow.com/>
>> 
>> 
>> 
> 
> Warmest Regards
>  
> Max Birley
> 
> Check this out
> 
> https://youtu.be/oGikNKcbIr <https://youtu.be/oGikNKcbIr>
> 
> Cell +1 830 998 9914
> 
> http://www.younomow.com/

Warmest Regards
 
Max Birley

Check this out

https://youtu.be/oGikNKcbIr

Cell +1 830 998 9914

http://www.younomow.com/

ROBOTIS-Will, 2022-02-03:
@Mdbirley 
Thank you for pasting the output messages.
The camera image related topics are not being published.
Do you get any error/warning messages on the terminal during the launch?

Mdbirley, 2022-02-03:
I will test again in the morning
from memory northing serious

> 
> This email originated from outside of 5 Hertford Street. Do not click on any links or open attachments unless you recognise the sender and know the content is safe.
> 
> 
> 
> 
> 
> 
> @Mdbirley <https://linkscan.io/scan/ux/aHR0cHM6Ly9naXRodWIuY29tL01kYmlybGV5/3754A46C6CF10C6E6FD7479D2FF5B1FFF47B39562A81665A921475815CE26C6A?c=5&i=1&docs=1>
> Thank you for pasting the output messages.
> The camera image related topics are not being published.
> Do you get any error/warning messages on the terminal during the launch?

Warmest Regards
 
Max Birley

Check this out
https://youtu.be/oGikNKcbI0

Please respond to all emails from ***@***.*** and reply to  ***@***.*** as ***@***.*** will not be operable by late November

Cell +1 830 998 9914

Our Web Site is you no mow
http://www.younomow.com/

Mdbirley, 2022-03-12:
Please can you answer this email request

> On 10 Mar 2022, at 21:44, ***@***.*** wrote:
> 
> 
> Hi
> I have a turtlebot 3 and I have just made another ros robot to see for myself what is involved.
> 
> The turtlebot 3 I manage through my ubuntu laptop running 16.04
> All is working well. I run the code roscore on the PC
> then bringup on the raspberry pi and then, to save raspberry pi processing power I run RVIZ on the laptop.
> 
> I have been controlling the second robot “Mybot-1” by vnc into the raspberry pi 
> however as the apple computer does not have ros installed I have to run all the nodes on the raspberry pi.
> This includes heavy memory users like rviz
> 
> I tried to set up the ~/.bashrc file on the Mybot-1 to be the same as the turtlebot 3 but I still can not see all the nodes or see data
> 
> Can you please provide me with the setup instruction if I want to manage either of the 2 rovers from the same laptop.
> I do not need to run them at the same time. I just want to offload the Rviz onto the laptop so the raspberry pi does not get overwhelmed
> 
I would prefer rescore to be run on the remote micro processor

> I guess that this would be the same as if  I want to run two turtlebots from the same laptop (at different times)
> 
> Many thanks Regards
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 

Warmest Regards
 
Max Birley

Check this out
https://youtu.be/oGikNKcbI0

Please respond to all emails from ***@***.*** and reply to  ***@***.*** as ***@***.*** will not be operable by late November

Cell +1 830 998 9914

Our Web Site is you no mow
http://www.younomow.com/

Mdbirley, 2022-10-11:
Hi
I have a turtlebot 3 and I have just made another ros robot to see for myself what is involved.

The turtlebot 3 I manage through my ubuntu laptop running 16.04
All is working well. I rune the code core on the PC
then bringup on the raspberry pi and then to save raspberry pi processing power I run RVIZ on the laptop.

I have been controlling the second robot “Mybot-1” by vnc into the raspberry pi 
however as the apple computer does not have ros installed I have to run all the nodes on the raspberry pi.
This includes heavy pricing power users like rviz

I tried to set up the ~/.bashrc file on the Mybot-1 to be the same as the turtle boy 3 but I still can not see all the nodes 

Can you please provide me with the setup instruction if I want to manage either of the 2 rovers from the same laptop.
I do not need to run them at the same time. I just want to offload the Rviz onto the laptop so the raspberry pi does not get overwhelmed

I guess that this would be the same as if  I want to run two turtlebots from the same laptop (at different times)

Many thanks Regards

GyuH13, 2024-10-31:
This issue was raised so long ago that it is now challenging to address.  
If you have any further questions, feel free to open a new issue

closed 2024-11-08 by sunghowoo