ROBOTIS-GIT/turtlebot3 #803: build tutlebot3 on ROS2, build error

yuan-ge, 2021-11-24:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [*] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [*] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [*] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [*] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [*] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [2.1.3]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    -  mkdir -p ~/tb3_ws/src
    -  cd ~/tb3_ws
    -  wget https://raw.githubusercontent.com/ROBOTIS-GIT/turtlebot3/ros2/turtlebot3.repos
    - vcs import src < turtlebot3.repos
    - colcon build --symlink-install

8. Copy and Paste the error messages on terminal.

Starting >>> turtlebot3_msgs
Starting >>> dynamixel_sdk
Starting >>> hls_lfcd_lds_driver
Starting >>> turtlebot3_description
Finished <<< turtlebot3_description [0.33s]
Starting >>> dynamixel_sdk_custom_interfaces
Finished <<< dynamixel_sdk [0.78s]    
Starting >>> turtlebot3_cartographer
--- stderr: turtlebot3_msgs           
CMake Error at /opt/ros/foxy/share/rosidl_generator_c/cmake/rosidl_generator_c_generate_interfaces.cmake:168 (find_package):
  By not providing "Findament_cmake_cppcheck.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_cppcheck", but CMake did not find one.

  Could not find a package configuration file provided by
  "ament_cmake_cppcheck" with any of the following names:

    ament_cmake_cppcheckConfig.cmake
    ament_cmake_cppcheck-config.cmake

  Add the installation prefix of "ament_cmake_cppcheck" to CMAKE_PREFIX_PATH
  or set "ament_cmake_cppcheck_DIR" to a directory containing one of the
  above files.  If "ament_cmake_cppcheck" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /opt/ros/foxy/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:45 (rosidl_generate_interfaces)

---
Failed   <<< turtlebot3_msgs [0.97s, exited with code 1]
Aborted  <<< turtlebot3_cartographer [0.87s]     
Aborted  <<< dynamixel_sdk_custom_interfaces [1.72s]
Aborted  <<< hls_lfcd_lds_driver [3.86s]

Summary: 2 packages finished [4.16s]
  1 package failed: turtlebot3_msgs
  3 packages aborted: dynamixel_sdk_custom_interfaces hls_lfcd_lds_driver turtlebot3_cartographer
  2 packages had stderr output: hls_lfcd_lds_driver turtlebot3_msgs
  10 packages not processed

9. Please describe the issue in detail.

yuan-ge, 2021-11-24:
I find the old similar issue, but the link in that issue is invalid.:(

ROBOTIS-Will, 2021-11-25:
Hi @yuan-ge 

When performing simulation on PC, you don't need to run the TurtleBot3 hardware.
Please refer to the eManual below and don't forget to select the ROS version at the top of the page.
https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#pc-setup

and the simulation instructions here
https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/

The Intel Joule had been discontinued a few years ago and not supported anymore so we won't be able to provide tech support for setting up with the latest version.

Thanks!

yuan-ge, 2021-11-25:
I just tried the tutorial in this page: [ ](https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#pc-setup). And I do the steps listed below:
[image]
but the result is the same:

> Starting >>> dynamixel_sdk
> Starting >>> turtlebot3_msgs
> Starting >>> turtlebot3_description
> Starting >>> dynamixel_sdk_custom_interfaces
> Finished <<< turtlebot3_description [0.43s]                                                                                                                                        
> Starting >>> turtlebot3_cartographer
> --- stderr: turtlebot3_msgs                                                                                                                            
> CMake Error at /opt/ros/foxy/share/rosidl_generator_c/cmake/rosidl_generator_c_generate_interfaces.cmake:168 (find_package):
>   By not providing "Findament_cmake_cppcheck.cmake" in CMAKE_MODULE_PATH this
>   project has asked CMake to find a package configuration file provided by
>   "ament_cmake_cppcheck", but CMake did not find one.
> 
>   Could not find a package configuration file provided by
>   "ament_cmake_cppcheck" with any of the following names:
> 
>     ament_cmake_cppcheckConfig.cmake
>     ament_cmake_cppcheck-config.cmake
> 
>   Add the installation prefix of "ament_cmake_cppcheck" to CMAKE_PREFIX_PATH
>   or set "ament_cmake_cppcheck_DIR" to a directory containing one of the
>   above files.  If "ament_cmake_cppcheck" provides a separate development
>   package or SDK, be sure it has been installed.
> Call Stack (most recent call first):
>   /opt/ros/foxy/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
>   /opt/ros/foxy/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
>   CMakeLists.txt:45 (rosidl_generate_interfaces)
> 
> 
> ---
> Failed   <<< turtlebot3_msgs [0.94s, exited with code 1]
> Aborted  <<< turtlebot3_cartographer [0.61s]                                                                                                                            
> Aborted  <<< dynamixel_sdk [1.23s]                                                                                             
> Aborted  <<< dynamixel_sdk_custom_interfaces [4.28s]                                 
> 
> Summary: 1 package finished [4.48s]
>   1 package failed: turtlebot3_msgs
>   3 packages aborted: dynamixel_sdk dynamixel_sdk_custom_interfaces turtlebot3_cartographer
>   1 package had stderr output: turtlebot3_msgs
>   7 packages not processed

yuan-ge, 2021-11-25:
so annoying. :(

yuan-ge, 2021-11-25:
Holy shit! I tried use the deb to install tutlebot3. When I run the app, I find my ros2 has no command "launch". So, I reinstall the foxy on my PC.  It's all right up to now.
The source of the problem is ROS2, my god.

closed 2021-12-08 by yuan-ge

1at7, 2021-12-16:
I was getting the same error but all of sudden I cleaned and rebuilt the workspace and its gone.!