ROBOTIS-GIT/turtlebot3 #804: Raspicam and ROS Melodic

DoppiaG93, 2021-12-03:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

(Not appeared on bring up)

7. Specify the commands or instructions to reproduce the issue.

    - catkin_make

8. Copy and Paste the error messages on terminal.

    - Could not find mmal libraries.

9. Please describe the issue in detail.

    - I have installed on turtlebot3 ROS melodic with ubuntu 18.04.  I have tried to install and compile all of the turtlebot packages and everything works except for the camera package. So, I did some research about the camera package and I saw that it uses raspicam, which I could check on the github repository that this is only available for ROS Kinetic. I have tried to install it anyway but it gave me an error about the mmal package that is unavailable. I could not check any solution about this kind of error online, so I would like to ask you if you could give me a possible solution to compile the camera package on ROS melodic.

ashekim, 2021-12-06:
HI. @DoppiaG93 

raspicam_node is only supported up to the kinetic version.
To use the camera in melodic or higher versions, You need to build opencv yourself by referring to the link below.

[ Click to expand : Autorace Package Installation for an actual TurtleBot3](https://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#install-autorace-packages-1)

[image]

To install opencv, proceed from step 2.
Thank you

DoppiaG93, 2022-02-28:
Hi,

I have installed all the opencv nodes and I have tried it with a USB web cam and it works! But, unfortunately, I have checked that with the camera mounted on the robot it does not work. It seems that the turtlebot is not able to read the camera... How can I check if the camera is correctly connected? 

Thank you

ashekim, 2022-03-02:
Hi @DoppiaG93 

Enter the following command in the terminal.
`ls /dev/video*`

If /dev/video0 is displayed in the result, the camera is connected normally.
If /dev/video0 does not appear, it means that there is no connection.

Additionally,
In the config.txt file located in /boot/firmware on the Raspberry Pi
[image]

**add start_x=1 before the enable_uart=1 line.**
[image]

Thank you.

ricemilk456, 2022-04-12:
Hi,

i'm using Raspberry Pi 4B,
This is how I installed mmal library, maybe it will help you:
```
cd ~/catkin_ws/src
git clone https://github.com/6by9/userland.git
cd userland
git checkout 64bit_mmal
./buildme --aarch64
sudo cp -a ~/catkin_ws/src/userland/build/lib/. /opt/vc/lib/
sudo cp -r /opt/vc/. /usr
```

closed 2024-11-13 by sunghowoo