ROBOTIS-GIT/turtlebot3 #806: Incomplete tf tree in Gazebo simulation

rlkamalapurkar, 2021-12-07:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [X] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [X] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [X] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [X] Ubuntu 20.04 LTS (Focal Fossa)
    - [X] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - Launch the `RviZ` simulator
      - `roslaunch turtlebot3_fake turtlebot3_fake.launch`
    - Visualize the `tf` tree
      - `rosrun rqt_tf_tree rqt_tf_tree`
   
    - Launch the `Gazebo` simulator
      - `roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch`
    - Visualize the `tf` tree
      - `rosrun rqt_tf_tree rqt_tf_tree`

8. Copy and Paste the error messages on terminal.

    - none

9. Please describe the issue in detail.

    - The `RviZ` simulation publishes a complete `tf` tree.
    [image]

    - The `Gazebo` simulation only publishes `odom` and `base_footprint` frames.
    [image]

    - Is this intentional? What can I do to get the `Gazebo` simulation to publish the complete `tf` tree?

rlkamalapurkar, 2021-12-09:
I just found out that the `tf` tree published by the real TurtleBot3 is identical to the Gazebo version, just `odom` and `base_footprint`. Why doesn't the TurtleBot3 publish all the other transformations that the fake node does?

davidlai0522, 2022-09-09:
I'm not sure if this is what you mean but you can try running these 2 nodes and check the frame again. I am adding them into my launch file (written in xml format):
<node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
<node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />

A1exan10er, 2023-03-02:
Is it because you're using a 3rd party LIDAR? 

I met a similar problem in Rviz using ROS2 Foxy. In Rviz it showed several TF warnings that "no information was transfered from [...] to [map]".
Then I checked the code of lidar itself and found nothing related to the TF node. So it means robot doesn't know anything other than the 2D point clouds. 

Maybe you need to write this part by yourself.

yun-goon, 2024-11-12:
Hi, @rlkamalapurkar 
The turtlebot3_fake package and Gazebo-based simulation are different.

The turtlebot3_fake package is primarily designed for simple visualization in RViz, allowing you to view basic movement of the TurtleBot3 model without detailed physical interactions or sensor simulations. It provides a basic TF tree structure to show the robot model's movement in RViz, but it doesn’t replicate the full physical environment.

On the other hand, Gazebo is a more advanced simulation platform that includes a physics engine, allowing for realistic interactions and sensor data simulation. When running a TurtleBot3 simulation in Gazebo, you get a TF tree that matches the real robot's structure, unlike the simpler TF tree from turtlebot3_fake.

sunghowoo, 2024-12-06:
@rlkamalapurkar This issue will be closed due to prolonged inactivity. Please try again with the latest versions, and if the issue persists, feel free to reopen it.

closed 2024-12-06 by sunghowoo