ROBOTIS-GIT/turtlebot3 #808: rViz simulation size on Windows

rlkamalapurkar, 2021-12-08:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [X] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [X] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [X] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [X] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - Launch rViz simulation using `roslaunch turtlebot3_fake turtlebot3_fake.launch`

8. Copy and Paste the error messages on terminal.

[image]

9. Please describe the issue in detail.

    - The simulation display starts in a tiny rectangle at the bottom left corner of the window (see image above). It resizes and takes up the whole window, as it should from the beginning, as soon as I resize the rViz window.

ashekim, 2021-12-10:
Hello @rlkamalapurkar.
First of all, it would be appreciated if you could understand that ROS window is not an official supported OS for TurtleBot3.

I think there is a problem with Rviz.
If you post it on the [ros forum](https://answers.ros.org/questions/) or [rviz github issue](https://github.com/ros-visualization/rviz/issues), you will be able to receive more accurate feedback.

thank you

sunghowoo, 2024-11-13:
This issue will be closed due to prolonged inactivity. if the issue persists, feel free to reopen it.

closed 2024-11-13 by sunghowoo