ROBOTIS-GIT/turtlebot3 #809: TELEOPERATION PROBLEM

Lopetex, 2021-12-09:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [X] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [X] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [X] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [X] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - ros2 run turtlebot3_teleop teleop_keyboard

8. Copy and Paste the error messages on terminal.

    - 

9. Please describe the issue in detail.

    - Everything installed following the emanual guide of TurtleBot3 burger (FOXY)
    - I ssh into my raspberry from my remote PC, bringup ($ ros2 launch turtlebot3_bringup robot.launch.py) and everything perfect.
    - When I try to control the robot with "wasd" I can see on the screen the linear/angular velocity is changing, but the robot is not moving. In Gazebo I don't have this problem.
    - I know I'm missing something because I also found the following problems: 
    - (1) when I rqt < topic monitor I only have 2 topics (/rosout and /paremter_events), also if I "$ ros2 topic list" the same 2. Service List is empty.
    - (2) when I run SLAM Node, the map is empty (I don't have my turtlebot on the map and I cannot find any tutorial or how to do it) and on terminal this message keeps pinging:
    [occupancy_grid_node-2] [WARN] [1639071108.423747834] [occupancy_grid_node]: submap slices and last_frame_id is empty

In case template fails:
Burger < ROS2 FOXY < Raspberry pi 3B+ < Raspbian by robotis < Ubuntu 20.04 LTS

ashekim, 2021-12-10:
Hi. @Lopetex 
ROS Domain ID Setting In ROS2 DDS communication, 
ROS_DOMAIN_ID must be matched between Remote PC and TurtleBot3 for communication under the same network environment.

Enter the `nano ~/.bashrc` command, Please check if the ROS_DOMAIN_ID of TurtleBot3 and Remote PC is the same.

[image]

mvgianel, 2021-12-10:
Hello,

I have the same problem in Waffle_pi with Raspberry pi 4. I have the ROS_DOMAIN_ID set in both the Turtlebot and the remote PC but I still have the same error. 

I noticed that in the e-manual these topics are present with bringup:
[image]

But I only see these topics:
[image]

It seems the rcl interfaces topics are not being published. Am I doing something else wrong?

Lopetex, 2021-12-10:
Hello @ashekim, checked this morning for the ROS_DOMAIN_ID, they're both set on 30.
Here I attach you the log of the cmd when I do the bringup:

ubuntu@ubuntu:~$ ros2 launch turtlebot3_bringup robot.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2021-12-09-20-09-50-841117-ubuntu-3270
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
/opt/ros/foxy/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/opt/ros/foxy/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
[INFO] [robot_state_publisher-1]: process started with pid [3272]
[INFO] [hlds_laser_publisher-2]: process started with pid [3274]
[INFO] [turtlebot3_ros-3]: process started with pid [3276]
[hlds_laser_publisher-2] [INFO] [1639080592.882198221] [hlds_laser_publisher]: Init hlds_laser_publisher Node Main
[hlds_laser_publisher-2] [INFO] [1639080592.883166663] [hlds_laser_publisher]: port : /dev/ttyUSB0 frame_id : base_scan
[turtlebot3_ros-3] [INFO] [1639080592.908473585] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [1639080592.912783498] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-3] [INFO] [1639080592.919547744] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [INFO] [1639080592.930102785] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link base_link had 5 children
[robot_state_publisher-1] Link caster_back_link had 0 children
[robot_state_publisher-1] Link imu_link had 0 children
[robot_state_publisher-1] Link base_scan had 0 children
[robot_state_publisher-1] Link wheel_left_link had 0 children
[robot_state_publisher-1] Link wheel_right_link had 0 children
[robot_state_publisher-1] [INFO] [1639080592.958330344] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1639080592.958867326] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1639080592.958962274] [robot_state_publisher]: got segment base_scan

There are some UserWarnings at the beginning I didn't notice and may have some relation on it.
Thanks for your quick response

Lopetex, 2021-12-10:
[image]

ashekim, 2021-12-13:
@Lopetex 
Hi again :)

```
/opt/ros/foxy/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
Node(
/opt/ros/foxy/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
```
This is just a deprecation warning. So it has nothing to do with the Teleoperation issue.

Excuse me, but could you please check if the Remote PC and TurtleBot3 are connected with the same wifi?
I'd appreciate it if you check it and let me know the results

Thank you.

ashekim, 2021-12-13:
HI. @mvgianel 
your topic list is working fine. The image below is my rqt topic monitor

[image]

Please refer to the [link](https://design.ros2.org/articles/intraprocess_communications.html) for information related to `topic/_instra`

Excuse me, but could you please check if the Remote PC and TurtleBot3 are connected with the same wifi?
I'd appreciate it if you check it and let me know the results.

Thank you.

Lopetex, 2021-12-13:
Hi @ashekim
They're both on the same network, raspberry is connected through wifi and my virtualmachine through ethernet. Is there is any problem with that? I don't have a wifi card on my PC so I cannot make the virtualmachine connect through wifi.
I attach you some screenshots:
[image]
[image]

mvgianel, 2021-12-13:
Hello @ashekim ,

Thank you for your prompt response. I will take a look at the link you provided me to understand the topics better.

I checked if my PC and the Turtlebot3 were connected to the same WiFi and they are, as shown here:
[image]
Also, in case this is important, my PC is not a Virtual machine. I have a Windows/Ubuntu dual boot.

I also launched the turtlebot3_bringup from the Raspberry Pi and tried moving the robot with the included remote control, and the robot didn't move. I tested the motors with Arduino, so I know they work. However, after I upload the firmware for the OpenCR board through Arduino I get a "Failed connection with devices" error when I launch the bringup, and when I upload the firmware through the Pi, I don't get the error but I encounter the same problem as @Lopetex.

ashekim, 2021-12-14:
@Lopetex  @mvgianel 
Thank you for your kind response.
If the network and domain are set up normally,
It is recommended to update or recover the opencr firmware.
If it doesn't work after that, please feel free to tell me.
thank you.

Lopetex, 2021-12-14:
Hi @ashekim
I've updated the firmware of the OpenCR this morning, no errors everything looks ok:
[image]
Bringup fine too, when I try the teleoperation it doesn't move.
I'm looking forward to your response, thanks for the support!

mvgianel, 2021-12-15:
Hi @ashekim and @Lopetex ,

I updated the firmware to the latest version and encountered the same problem. Even in recovery mode, the latest version did not allow me to teleop the robot. However, I booted the OpenCR board in recovery mode and uploaded the firmware version 0.0.2, and that allowed me to launch the bringup and teleop the robot with the keyboard and with the control. The only problem is that when I turn off the control, the robot starts moving by itself.

ashekim, 2021-12-15:
Hi ALL :)

@Lopetex  
If the Opencr upload was successful, my guess is the network settings in the VM environment.
I searched for VM Network Setting method. You may want to refer to the solution below.
[VirtualBox and Turtlebot 3. During RVIZ, unable to contact my own server at 10.0.2.15](https://discourse.ros.org/t/virtualbox-and-turtlebot-3-during-rviz-unable-to-contact-my-own-server-at-10-0-2-15/3859)

@mvgianel
First of all, congratulations on the successful TB3 Teleoperation.
Does turn off the control mean terminate the teleoperation?

Thank you both for your replies. 
I hope you both have a good day :)

Lopetex, 2021-12-15:
Hi! @mvgianel
According to you suggest, I did the following on the raspberry:
 $ sudo dpkg --add-architecture armhf
 $ sudo apt update
 $ sudo apt install libc6:armhf

$ export OPENCR_PORT=/dev/ttyACM0
$ export OPENCR_MODEL=burger
$ rm -rf ./opencr_update.tar.bz2

**$ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/0.0.2/opencr_update.tar.bz2**
$ tar -xjf ./opencr_update.tar.bz2
$ cd ~/opencr_update
$ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr

And then the command for the bringup on the Raspberry and on the PC the teleoperation.
So I did that, and it's still not working...
I'm I supposed to delete somehow first the previous firmware I've downloaded? In that case, could you tell me how?
Thanks for being involved in this and your support.

Lopetex, 2021-12-15:
@ashekim 
Thanks again for your support, I'm going to give you some information of my VM environment and so the raspberry and PC.
1. PC -> Windows 10 using Virtual Box for the virtual Machine. **IP: 192.xxx.xxx.138**
2. VM -> Virtual Box is connected to my PC through a Bridge adapter of Ethernet because I don't have a Wi-Fi card on my PC. Is this the problem? Because there are people claiming it may be. **IP: 192.xxx.xxx.130**
3. Raspberry -> I'm using the RPi3B+ ROS2 Foxy image that the guide provides. I've noticed that the IP of this one sometimes changes (it's random, I've been 2 days with the same IP and then suddenly changed). Started being: 192.xxx.xxx.130 (the same as the VM) and swapped to 192.xxx.xxx.134 and 192.xxx.xxx.135 and so on. So every time I want to connect, I first plug the raspberry to another screen and do "ifconfig" to make sure it didn't change

I'm sure this could be the main issue of the thing, according to the link you posted me, I don't think it fits my situation. If setting a static IP for the Raspberry solves it, which one should it be, the same as the VM or a random one?

Last thing, my final thesis is related with working with the turtlebot3 and I need technical support really asap, do you know how I can contact to ROBOTIS or support department to get remote assisted or maybe through phone?
Thanks a lot for being involved.

ROBOTIS-Will, 2021-12-20:
Hi @Lopetex 

Please make sure that you have correctly installed the right package.
The correct Foxy package does not contain any code related to the warnings below.

> /opt/ros/foxy/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
Node(
/opt/ros/foxy/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead

Also please note that we cannot provide in-depth tech support for the virtual machine environment as there are way too many variations that we can cover.
Thank you for your understanding.

Lopetex, 2022-01-16:
@ROBOTIS-Will 

Good morning, thanks for support.
I've spent this month trying to figure out what was wrong and following your advice, I've tried to redo it again everything.
The problem is still there.

As ashekim said:

> This is just a deprecation warning. So it has nothing to do with the Teleoperation issue.
> 
> Excuse me, but could you please check if the Remote PC and TurtleBot3 are connected with the same wifi? I'd appreciate it if you check it and let me know the results
> 
> Thank you.

I assume that my issue is related to how I connect to the robot and how the ROS_DOMAIN_ID works.
As I said before on this issue, I'm using a PC, ethernet connection and giving internet access to my virtual machine through bridged connection of Ethernet. My raspberry though was connected by Wi-Fi.
Quite sure both (virtual machine and raspberry) need to work through Wi-Fi.
Bought a wireless adapter for my pc and succeed on making my virtual machine work with Wi-Fi. 

Making new tests to try to solve the problem.

If this is not working, I want to swap to NOETIC. **My question is, how can I clean the OpenCR1.0?**
Now it has all the FOXY files in it.

Thanks for the help.
- Pol

ROBOTIS-Will, 2022-01-17:
Hi @Lopetex 

Thanks for the updates.
As Ashe mentioned above, the deprecation warnings are caused by the deprecation of name of some parameters, but they are safe to run.
It seems some of these changes has not been applied to the LDS Foxy package.
I'll review and update in the next cycle (which will be this month).

Usually(I mean non virtual machine environment), as long as the Raspberry Pi on TurtleBot3 and your PC is in the same local network, they don't necessarily have to be connected via WiFi.
You may keep your PC connected to the router via Ethernet, while connecting the TurtleBot via WiFi.

In order to use the Noetic version, you may follow the OpenCR Setup guide of the TurtleBot3 eManual below.
Please make sure to select the `Noetic` by clicking the `Noetic` button at the top of the page, or else you'll see the Kinetic instructions.
https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup

Thank you.

Lopetex, 2022-01-17:
Hi @ROBOTIS-Will

Thanks for your response.
 Yes, I'm aware that the guide has different buttons on top of the page and can generate some issues to users sometimes.

Why I'm assuming the Wi-Fi connection:
As you might know, ROS2 uses the ROS_DOMAIN_ID as the communication sistem (DDS) on this particular one, I cannot tell if it's mandatory or not.

With the ROS1 communication,
[image]

On the guide itself, it specifies you to connect to a Wi-Fi device, even on the following image, they choose the wlp2s0 IP instead of the enp3s0.

[image]

Leaving that aside, my problem is the following:
As FOXY couldn't work for me, I made a new virtual machine with Noetic Ninjemis.
Burned the Raspberry Pi 3B+ ROS Noetic image with Disks and resized with gparted.
**Is any way to restore the OpenCR1.0 from factory?** It has all the previous firmware of the FOXY version and is causing some issues while booting the Raspberry.

Thanks for the support!

Lopetex, 2022-01-18:
Hi, I'll attach some pictures of the bashrc script when I boot the raspberry:
[image]
[image]

As you can see, there's no ROS_MASTER_URI nor ROS_HOSTNAME and you can still see there's FOXY information.

Lopetex, 2022-01-20:
As long as I'm not working with ROS2 FOXY I'll set this issue as unsolved.

In case anyone is reading this issue seeking for information and encountered a situation similar to mine i suggest to swap to Noetic instead of Foxy.

Thanks everyone for the support.

closed 2022-01-20 by Lopetex

ROBOTIS-Will, 2022-01-24:
Hi @Lopetex 
Sorry about not being much helpful for your virtual environment.
The DDS is a default middleware to operate ROS2, and it doesn't necessarily have to be a WiFi.
Some routers may not compatible with DDS, but we do not have much experience / information in this area
Although the eManual is written with WiFi connection which is probably the most preferred connection for a mobile robot, you may also use ethernet connection if your robot is stationary like Manipulators.
Thank you!