ROBOTIS-GIT/turtlebot3 #811: Pi camera issues for ROS kinetic on Pi 3b+

stevensu1838, 2021-12-24:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.
Issue 1 : Camera is not detected.
pi@raspberrypi:~ $ raspistill -v -o test.jpg
mmal: Cannot read camera info, keeping the defaults for OV5647

"raspistill" Camera App (commit 7cbfbd38d982 Tainted)

Camera Name OV5647
Width 2592, Height 1944, filename test.jpg
Using camera 0, sensor mode 0

GPS output Disabled

Quality 85, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Time delay 5000, Timelapse 0
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Flicker Avoid Mode 'off'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

pi@raspberrypi:~ $ 

Issue 2 : Fail to install relevant Raspberry Pi Camera 
pi@raspberrypi:~ $ sudo apt-get install ros-kinetic-compressed-image-transport ros-kinetic-camera-info-manager
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-compressed-image-transport
E: Unable to locate package ros-kinetic-camera-info-manager
pi@raspberrypi:~ $ 

8. Copy and Paste the error messages on terminal.
See above

9. Please describe the issue in detail.
After I installed the Raspbian ROS Kinetic OS for P 3b+, I did the following

Once logged in to the Raspberry Pi, execute the commands below to sync time.
$ sudo apt-get install ntpdate
$ sudo ntpdate ntp.ubuntu.com
Load Raspberry Pi configuration interface.
$ sudo raspi-config
Select Advanced Options > Expand Filesystem and exit.
 
And then I followed the following official document to activate and run the RPi camera, but no success. 
[Turtlebot3 Pi Camera Setup](https://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_raspi_cam/#raspberry-pi-camera-for-turtlebot3) 
  
I can make sure the Pi camera is connected right and I've checked with different Pi 3b+ boards. same issues. Thanks a lot

stevensu1838, 2021-12-26:
It turns out I can use the Pi camera without installing the following two package:

E: Unable to locate package ros-kinetic-compressed-image-transport
E: Unable to locate package ros-kinetic-camera-info-manager

When I run $ roslaunch turtlebot3_bringup turtlebot3_rpicamera.launch

Only a Warn message shows up and I can visualize the image on my remote PC. 

[ WARN] [1640498104.677588643]: Camera calibration file /home/pi/.ros/camera_info/camera.yaml not found.

The whole printout is shown as below:
pi@raspberrypi:~ $ roslaunch turtlebot3_bringup turtlebot3_rpicamera.launch
... logging to /home/pi/.ros/log/fef183dc-654c-11ec-a6ad-480fcf41d72a/roslaunch-raspberrypi-12191.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.108:38719/

SUMMARY
========

PARAMETERS
 * /raspicam_node/camera_frame_id: camera
 * /raspicam_node/camera_info_url: package://turtleb...
 * /raspicam_node/framerate: 15
 * /raspicam_node/height: 480
 * /raspicam_node/width: 640
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    raspicam_node (raspicam_node/raspicam_node)

ROS_MASTER_URI=http://192.168.0.104:11311

process[raspicam_node-1]: started with pid [12200]
[ INFO] [1640498104.496394676]: splitter component done

[ INFO] [1640498104.603253641]: camera calibration URL: package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml
[ INFO] [1640498104.675481157]: Camera successfully calibrated from default file
[ INFO] [1640498104.676131205]: using default calibration URL
[ INFO] [1640498104.676628285]: camera calibration URL: file:///home/pi/.ros/camera_info/camera.yaml
[ INFO] [1640498104.677214427]: Unable to open camera calibration file [/home/pi/.ros/camera_info/camera.yaml]
[ WARN] [1640498104.677588643]: Camera calibration file /home/pi/.ros/camera_info/camera.yaml not found.
[ INFO] [1640498104.677889735]: No device specifc calibration found
[ INFO] [1640498104.819591295]: Starting video capture (640, 480, 80, 15)

[ INFO] [1640498104.821371961]: Video capture started

ROBOTIS-Will, 2021-12-29:
Hi @stevensu1838 

Yes, the images are preinstalled with necessary packages to run TurtleBot3.
However, the provided Kinetic image is created before the ROS GPG Key incident, therefore, when trying to updating the package or installing a new package with `apt-get update` or `apt-get upgrade` commands will return such errors.
Please see below thread for more details and resolutions.
https://discourse.ros.org/t/ros-gpg-key-expiration-incident/20669/38

Thanks!

closed 2024-11-13 by sunghowoo