ROBOTIS-GIT/turtlebot3 #813: turtlebot3_teleop does not repeat messages on Windows

rlkamalapurkar, 2022-01-10:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [X] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [X] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [X] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [X] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: Not Relevant
    - Firmware version: Not relecant

7. Specify the commands or instructions to reproduce the issue.

    - On the Windows 10 PC, open a ROS terminal and run `roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch` 
    - On the Windows 10 PC, open a new ROS terminal and run `rostopic echo /cmd_vel`

8. Copy and Paste the error messages on terminal.

    - No error message

9. Please describe the issue in detail.

    - When I press, for example, `w` to send a linear velocity command, a corresponding `twist` message is echoed in the `echo` terminal *only once*. The message is not repeated. As a consequence, the TurtleBot moves a little and stops.
    - On a Linux machine, the same `teleop` node repeats the `twist` message every 0.1s, so the robot moves as it should.

ashekim, 2022-01-14:
Hello @rlkamalapurkar 
Thanks for asking and sorry for the late reply

e-manual was tested on Intel NUC with windows installed instead of raspberry pi (SBC).
`This SBC Setup section for Windows is tested with Intel Up² or Intel NUC on TurtleBot3.`
Unfortunately, there is no test support in Remote PC windows & SBC raspberry pi environment, 
so it is difficult to accurately support.

Therefore, I am attaching a good [link](https://answers.ros.org/question/50507/ros-on-ubuntu-communicating-with-win_ros/) for your reference.
Please check your windows firewall

In addition, If you contact the ROS Forum below, you will receive a quick and accurate answer.
[ROS Forum](https://answers.ros.org/questions/)

Thanks for your understanding.

rlkamalapurkar, 2022-01-14:
Thanks for the response. The SBC is irrelevant, though, you do not even need a TurtleBot3 to reproduce this issue.

The turtlebot3_teleop package, when installed on Windows 10, does not repeat messages. On Linux, it does. Any ideas that will help me understand why would be appreciated!

rlkamalapurkar, 2022-01-16:
@ashekim I found the problem, but I am not sure how to fix it. The Windows implementation of the `getKey` function in [the python code](https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_teleop/nodes/turtlebot3_teleop_key) is this:
```
    if os.name == 'nt':
        if sys.version_info[0] >= 3:
            return msvcrt.getch().decode()
        else:
            return msvcrt.getch()
```
The problem is that `getch()` is blocking. So, unless a key is pressed on the keyboard, the execution gets stuck inside `getKey` and the while loop on line 144 only publishes once per keypress. 

The Linux implementation includes a 0.1 second timeout, the windows implementation does not. I tried
```
    if os.name == 'nt':
        timeout = 0.1
        startTime = time.time()
        while(1):
            if msvcrt.kbhit():
                print("""keypress""")
                if sys.version_info[0] >= 3:
                    return msvcrt.getch().decode()
                else:
                    return msvcrt.getch()
            elif time.time() - startTime > timeout:
                return ''
```
but unfortunately, for some reason, `msvcrt.kbhit()` is unable to read `Ctrl` key. Any ideas on how a timeout for `getch()` can be added to the Windows implementation?

ashekim, 2022-01-26:
Hi @rlkamalapurkar.
Thank you for your contribution.
Your PR will be tested and merged into the develop branch.
Once again, thank you for your help and supports.

Have a good day! :)

closed 2022-01-26 by ashekim