ROBOTIS-GIT/turtlebot3 #816: Turtlebot3 detects Fake Obstacles

DistroIII, 2022-01-18:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ yes] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ Melodic] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [Yes ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ Yes] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ Yes] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    - The LiDAR on turtlebot3 is detecting fake obstacles where there is free space. I am unsure as to what is happening. I have no idea if this is a hardware issue or something else. Attached are two screenshots of what my robot senses in Rviz and what is actually going on in the Real World respectively. As clearly seen, there is free space around the robot yet it detects obstacles all around it.
    
/home/philip/Desktop/Screenshot from 2022-01-17 18-14-15.png

/home/philip/Desktop/2022-01-17-181946.jpg

ashekim, 2022-01-24:
Hello. @DistroIII 
The laser filters package can be used to filter laser scans.
I attached related [references](http://wiki.ros.org/laser_filters).
Thank you

DistroIII, 2022-01-26:
@ashekim  I have benn trying to figure out how to use the laser filter package but have not been able to get it to work. I need help please. here is the launch file I'm trying to use to run it.

```
<launch>
  <node pkg="laser_filters" type="scan_to_scan_filter_chain"
      name="laser_filter">
    <rosparam command="load" file="$(find test)/my_laser_config.yaml" />
    <remap from="scan" to="base_scan" />
  </node>
</launch>
```

I know that there is something going on with the "  <remap from="scan" to="base_scan" />", as turtlebot3 uses the \scan topic and not the \base_scan, but when I remove this line I simply get an error when I roalaunch. I'm at my wits end, Any help would be greatly appreaciated.

ROBOTIS-Will, 2022-02-03:
@DistroIII 
This could be the environmental problem where there are some reflective materials in the room.
Or have you cleaned up the window of the LDS sensor?
Try gently cleaning the sensor using a microfiber cloth to remove any contamination.

closed 2024-11-13 by sunghowoo