ROBOTIS-GIT/turtlebot3 #817: Simulation fails after robot spawns

tiffanyyk, 2022-01-20:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

`roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch`

8. Copy and Paste the error messages on terminal.
```
auto-starting new master
process[master]: started with pid [196] 
ROS_MASTER_URI=http://localhost:11311 

setting /run_id to 825848e2-79a6-11ec-a24b-0242ac110002 
process[rosout-1]: started with pid [209] 
started core service [/rosout] 
process[gazebo-2]: started with pid [217] 
[ INFO] [1642651661.588953100]: Finished loading Gazebo ROS API Plugin. 
[ INFO] [1642651661.590139100]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting... process[gazebo_gui-3]: started with pid [237]
[ INFO] [1642651661.813988900]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1642651661.815237400]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting... process[spawn_urdf-4]: started with pid [309] 
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update' 
[ INFO] [1642651662.964731900]: waitForService: Service [/gazebo/set_physics_properties] is now available. 
[ INFO] [1642651662.977583500]: waitForService: Service [/gazebo/set_physics_properties] is now available. 
[ INFO] [1642651663.069240800, 0.065000000]: Physics dynamic reconfigure ready. 
[ INFO] [1642651663.071203500]: Physics dynamic reconfigure ready. 
[ INFO] [1642651663.804367900, 0.326000000]: Camera Plugin: Using the 'robotNamespace' param: '/' 
[ INFO] [1642651663.807579800, 0.326000000]: Camera Plugin (ns = /) <tf_prefix_>, set to ""
[ INFO] [1642651663.891404500, 0.326000000]: Laser Plugin: Using the 'robotNamespace' param: '/' 
[ INFO] [1642651663.891529100, 0.326000000]: Starting Laser Plugin (ns = /) 
[ INFO] [1642651663.897121000, 0.326000000]: Laser Plugin (ns = /) <tf_prefix_>, set to "" 
[ INFO] [1642651663.904435500, 0.326000000]: Starting plugin DiffDrive(ns = //) 
[ INFO] [1642651663.907123000, 0.326000000]: DiffDrive(ns = //): <rosDebugLevel> = na 
[ INFO] [1642651663.913773000, 0.326000000]: DiffDrive(ns = //): <tf_prefix> = 
[ INFO] [1642651663.917727900, 0.326000000]: DiffDrive(ns = //): Advertise joint_states 
[ INFO] [1642651663.922200800, 0.326000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel 
[ INFO] [1642651663.933103200, 0.326000000]: DiffDrive(ns = //): Subscribe to cmd_vel 
[ INFO] [1642651663.936081200, 0.326000000]: DiffDrive(ns = //): Advertise odom on odom 
[spawn_urdf-4] process has finished cleanly 
log file: /home/tiffanyyk/.ros/log/825848e2-79a6-11ec-a24b-0242ac110002/spawn_urdf-4*.log 
```
9. Please describe the issue in detail.

I'm trying to run the turtlebot teleop example by first launching the turtlebot in the empty world with `roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch` and then `roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch`. Unfortunately there seems to be problems right after the turtlebot model is spawned. I've attached my log files below.

[spawn_urdf-4-stdout.log](https://github.com/ROBOTIS-GIT/turtlebot3/files/7906414/spawn_urdf-4-stdout.log)
[spawn_urdf-4.log](https://github.com/ROBOTIS-GIT/turtlebot3/files/7906415/spawn_urdf-4.log)

In particular, `spawn_urdf-4.log` shows that there's some shutdown signal after the model is spawned. 

I'm running the simulation in a docker container on Windows 10 WSL2 with ROS Kinetic, Gazebo7.

ashekim, 2022-01-21:
Hi @tiffanyyk . 
Looking at ROS_MASTER_URI=http://localhost:11311 , I think it is a network configuration problem.

The Ubuntu app doesn't know the Windows host, 
so you need to create a connection from the Ubuntu app to your Windows laptop.
Enter the `ipconfig` command to find out the IP address of your Windows system.

And run the following command in ubuntu terminal.
`'export DISPLAY={your_ip_address}:0.0' >> ~/.bashrc`

Also, please note that we cannot provide in-depth tech support for the virtual machine environment as there are way too many variations that we can cover.

Therefore, you can get a more accurate answer by contacting the [ros forum](https://answers.ros.org/questions/).
Thank you for your understanding.

ashekim, 2022-01-21:
Hi @tiffanyyk .
Looking at ROS_MASTER_URI=http://localhost:11311 , I think it is a network configuration problem.

The Ubuntu app doesn't know the Windows host,
so you need to create a connection from the Ubuntu app to your Windows laptop.
Enter the ipconfig command to find out the IP address of your Windows system.

And run the following command in ubuntu terminal.
'export DISPLAY={your_ip_address}:0.0' >> ~/.bashrc

Also, please note that we cannot provide in-depth tech support for the virtual machine environment as there are way too many variations that we can cover.

Therefore, you can get a more accurate answer by contacting the ros forum.
Thank you for your understanding.

tiffanyyk, 2022-01-21:
Hi @ashekim ,
Thank you for your response. However, I had already set the `DISPLAY` variable in the same way you mentioned. Could there be anything unrelated to configuration and VM environment that is causing this problem? Thanks in advance.

tiffanyyk, 2022-02-02:
I'm going to close this issue now. It seems that this terminal output was unrelated to the simulation not working. Increasing memory allocation for docker made the simulation work for me, despite seeing the same terminal output.

closed 2022-02-02 by tiffanyyk