ROBOTIS-GIT/turtlebot3 #818: `rviz2` in `turtlebot3_bringup` really required?

goekce, 2022-01-21:
I am compiling TB3 packages for RPi. When I generate the dependencies as follows, then `rviz2` is also included as a dependency.
```
$ ROS_DISTRO=galactic ROS_PACKAGE_PATH=/opt/ros2/galactic/ rosinstall_generator turtlebot3 --deps --exclude RPP turtlebot3_navigation2 turtlebot3_cartographer
...
- git:                                                                                                                                                                                                                                       
    local-name: rviz/rviz2                                                                                                                                                                                                                   
...
```

It seems `turtlebot3_bringup` requires it:
https://github.com/ROBOTIS-GIT/turtlebot3/blob/bb4524c09650ccc3f1c619c1b9527efcb6762503/turtlebot3_bringup/package.xml#L16-L21

Is `rviz2` really required to bring up TB3?

ROBOTIS-Will, 2022-01-24:
Hi @goekce 

During the bringup of the TurtleBot3 robot itself, RViz2 package is not necessary so you may delete the RViz2 related code and dependencies.

However, the RViz2 is required for most cases when running with the remote PC unless your TurtleBot3 doesn't need a remote PC to operate like [TurtleBot3 Autorace Challenge](https://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#autonomous-driving).

Thank you.

goekce, 2022-01-24:
Dear Will thank you for your thorough reply.

I suppose that `turtlebot_bringup` also contains a launch script which can launch rviz, that is why you included `rviz2`. Don't you think that most users will run `turtlebot_bringrup` on headless computers without graphical interface and run `rviz2` on their remotes?

Still I can also understand the inclusion, this way the package stays more generic.

closed 2022-01-24 by goekce

ROBOTIS-Will, 2022-01-25:
@goekce 
Thank you for the suggestions.
I agree with your thoughts.
As the performance of the recent SBC is getting improved over time, the headless configuration as a standalone robot will be the goal that we should pursue.
I'll try to come up with reasonable modification that can reduce the size of the necessary packages for the robot.
Thanks!