ROBOTIS-GIT/turtlebot3 #819: Turtlebot3 IMU and ODOM yaw drifts

lslabon, 2022-01-21:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ x] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

Hi!

My Yaw is drifting (odom).
To smooth out the values is not the perfect solution for me.

I have noticed, that when i upload an example sketch from turtlebot (Read Roll Pitch Yaw) the values are not drifting :(

I tried to read the serial debug, but did not manage to get the readings ...

What is the best way to debug the OPENCR Board?

please see  --> Turtlebot3 IMU and ODOM yaw drifts and is non-zero on Startup #655

ROBOTIS-Will, 2022-01-24:
Hi @lslabon 

The easiest way to debug the Arduino code is through the Serial port of the OpenCR.
By default, turtlebot3 sketch provides debugging mode by enabling the DEBUG definition in the below.
https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core_config.h#L69

This will print sensor values with a defined frequency below.
https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core_config.h#L51

The default serial port assigned for debugging is Serial4 which is `UART2` port of OpenCR (the one that right next to the micro USB port).
You may use any serial-usb interfaces like [LN-101](https://emanual.robotis.com/docs/en/parts/interface/ln-101/) to read debugging messages.

[image]

You may also read the `imu` topic from the Turtlebot3 using tools like RQT to see if there's a yaw drifting.
[image]

lslabon, 2022-01-31:
Hi! 

Thanks for your answer!

I bought a ln-101 and checked the readings ...
Unfortunately i got a drift on the THETA and on the IMU Z reading.

Do you have any advice? Like this it is hard to use the opencr board, because i get a drift like 10 degree within 5min...

Maybee an additional calibration?
The drift is always in the same "direction".
Greets

ROBOTIS-Will, 2022-02-03:
Hi @lslabon 

The drift of the cheap IMU sensor is inevitable and may require additional work(e.g. repetitive adjustment during the operation) to minimize it, but 10&deg; drift in 5 minutes seems a bit too much.

I've left my TB3 Burger(ROS2 Foxy) for about 5 hours on my desk and had some drift about 12&deg; (0.21 radian)
[image]

Does the drift occur when the OpenCR is disassembled from the robot and left sitting alone away from any metallic/ferromagnetic objects?
When running the Navigation or SLAM, the drift can be negligible as the LDS or AMCL compensates errors from the IMU and encoders.
The OpenCR library does not have an implementation of calibration for MPU9250 and since the MPU9250 is discontinued and replaced with ICM-20648(6 DOF without the magnetometer), unfortunately, "official" development for MPU9250 library is unlikely to happen.

You may want to refer to the Kris Winer's repositories below regarding the MPU sensor.
https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration
https://github.com/kriswiner/MPU9250/issues/333

Thank you.

closed 2024-11-13 by sunghowoo