ROBOTIS-GIT/turtlebot3 #820: Turtlebot3 LiDAR issue fix not working

DistroIII, 2022-01-23:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ yes] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ melodic] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [yes ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [yes ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ yes] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    I had recently reported an issue with my real life turtlebot3 LiDAR sensor and how it detects fake obstacles. I found a package that might help me called [laser_filter](http://wiki.ros.org/laser_filters). I have a hard time making it work.I install the package, make a yaml file and make a launch file, I run the launch file, it runs but doesnt seem to do anything. I think I have to somehow incorporate the launch file with one of the turtlebot3 luanhc files, I think I'm maybe supposed to run the launch file through the turtlebot3_navigation.launch file? I know that the amcl.launch file and move_base.launch files are run through the turtlebot3_naviation.launch file so I'm guessing thats what I should so with my _laser_filter.launch file? I am very grateful for any help.

ROBOTIS-Will, 2022-01-24:
Hi @DistroIII 
You may include the laser_filter launch command in the turtlebot3_navigation launch file.
The default turtlebot3 packages use the `scan` topic for basic operation, so you should change this topic name to whatever the published topic name you wrote in the laser_filter node.
If the laser_filter node correctly receives the `scan` topic as an input, you should be able to see the filtered scan message as a publisher in your ROS network.
Thanks!

sunghowoo, 2024-11-13:
Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. please feel free to resubmit it, and we will address it promptly.

closed 2024-11-13 by sunghowoo