ROBOTIS-GIT/turtlebot3 #821: Odom topic gives extreme large readings on coordinate y

ByLiZhao, 2022-01-28:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ y] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ y] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ y] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ y] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ y] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [2.1.2]
    - Firmware version: [0.0.3]

7. Specify the commands or instructions to reproduce the issue.

   ` ros2 launch turtlebot3_bringup robot.launch.py`

8. Copy and Paste the error messages on terminal.

    No error message

9. Please describe the issue in detail.
Sometimes, the /odom topic gives extreme large readings on /odom/pose/pose/position/y. For several times, the value is Inf.
This does not always happens. I tried the above bringup commands many times, about 1 out of 5 times, the error happens.

Below is a screenshot of rqt to show an example.
[image]

ByLiZhao, 2022-01-28:
I observed that if I reboot the Raspberry Pi, and run the command immediately after the robot is ready, the y coordinate of "/odom/pose/pose/position" is always Inf. If, however, I run some other programs that consume large memory space, the problem only appear occasionally. This seems like accessing uninitialized memory.

ROBOTIS-Will, 2022-02-03:
Hi @ByLiZhao 
Thank you for reporting the issue.
I'll see if I can reproduce this on my burger with Foxy and get back to you.
Thanks!

lucasromary, 2022-05-09:
Hi,

Same problem here, with "ros2 launch turtlebot3_bringup robot.launch.py".
I have also a Turtlebot burger with Foxy.

sudar1220, 2022-06-26:
I have the same problem with Turtlebot3 Waffle pi on Foxy. I just posted an issue a couple of days back.
Maybe you want to follow it, since its newer: https://github.com/ROBOTIS-GIT/turtlebot3/issues/880

Or did you guys fix the issue already?

ROBOTIS-Will, 2023-01-27:
An updated OpenCR ROS2 firmware V220127R1 (0.2.1) is released.
Please [download and upload the new firmware](https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup) and let me know if this doesn't fix the problem.
Thank you very much for your patience!

spolstra, 2023-10-10:
Hi,

I ran into the same issue where the turtlebot would have very large x and/or y positions. This happened on foxy with a rpi4 board. I fixed this by initializing the pose member in the `Odometry` class. See #750 for my solution.

closed 2024-11-13 by sunghowoo