ROBOTIS-GIT/turtlebot3 #822: Turtlebot3 RPi4 Creation of publisher failed: Checksum does not match:

PratikMahale02, 2022-02-03:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [*] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [*] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [*] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [*] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [*] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - roslaunch turtlebot3_bringup turtlebot3_robot.launch

8. Copy and Paste the error messages on terminal.

    - ... logging to /home/ubuntu/.ros/log/31cd8ab6-84a7-11ec-be49-3d5b899ee966/roslaunch-ubuntu-2287.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.0.0.65:41993/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.11
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://10.0.0.252:11311

process[turtlebot3_core-1]: started with pid [2302]
process[turtlebot3_lds-2]: started with pid [2303]
process[turtlebot3_diagnostics-3]: started with pid [2304]
[INFO] [1643873078.451008]: ROS Serial Python Node
[INFO] [1643873078.518356]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1643873080.653999]: Requesting topics...
[INFO] [1643873080.806145]: Note: publish buffer size is 1024 bytes
[INFO] [1643873080.818852]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1643873080.846689]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1643873080.926394]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1643873080.950291]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1643873080.992183]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1643873081.016379]: Setup publisher on joint_states [sensor_msgs/JointState]
[ERROR] [1643873081.029412]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1643873081.049614]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1643873081.121563]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1643873081.152309]: Note: subscribe buffer size is 1024 bytes
[INFO] [1643873081.165757]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1643873081.192232]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1643873081.225842]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1643873081.262946]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1643873083.952614]: Setup TF on Odometry [odom]
[INFO] [1643873083.964244]: Setup TF on IMU [imu_link]
[INFO] [1643873083.980309]: Setup TF on MagneticField [mag_link]
[INFO] [1643873083.993953]: Setup TF on JointState [base_link]
[INFO] [1643873084.016577]: --------------------------
[INFO] [1643873084.026219]: Connected to OpenCR board!
[INFO] [1643873084.036474]: This core(v1.2.6) is compatible with TB3 Burger
[INFO] [1643873084.048131]: --------------------------
[INFO] [1643873084.061076]: Start Calibration of Gyro
[INFO] [1643873086.515381]: Calibration End
[ERROR] [1643873086.615906]: Tried to publish before configured, topic id 131
[INFO] [1643873086.626797]: Requesting topics...
[ERROR] [1643873086.675009]: Tried to publish before configured, topic id 131
[INFO] [1643873086.683600]: Requesting topics...
[ERROR] [1643873086.720821]: Tried to publish before configured, topic id 131
[INFO] [1643873086.731762]: Requesting topics...
[ERROR] [1643873086.809200]: Tried to publish before configured, topic id 131
[INFO] [1643873086.818405]: Requesting topics...
[ERROR] [1643873086.856441]: Tried to publish before configured, topic id 131
[INFO] [1643873086.868340]: Requesting topics...
[INFO] [1643873086.921483]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1643873086.933119]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1643873086.944805]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1643873086.961623]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1643873086.980690]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1643873086.996997]: Setup publisher on joint_states [sensor_msgs/JointState]
[ERROR] [1643873087.018051]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770

9. Please describe the issue in detail.

    - I followed the steps here but was unable to solve the problem.

ROBOTIS-Will, 2022-02-03:
Hi @PratikMahale02 
Please refer to #768 
Thanks.

PratikMahale02, 2022-02-03:
Hey @ROBOTIS-Will ,

Thank you for your quick response. I should have clarified initially.
The solution in the link you sent me points out to following the steps here for Noetic,
https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup

I followed through these steps during the initial setup and I was able to test the motors.
I got the checksum error when I got to the step to do run the command, 
    'roslaunch turtlebot3_bringup turtlebot3_robot.launch'

I tried the solution here to update the SBC and update the code on the OpenCR through Arduino as in the solutions in,
#108 
#692 
Non of these helped.

What does the error mean?

PratikMahale02, 2022-02-05:
Hi @ROBOTIS-Will.
That worked. Not sure what I missed earlier!
Thank you!

closed 2022-02-05 by PratikMahale02