ROBOTIS-GIT/turtlebot3 #823: Avoid obstacles while running turtlebot_follower

IsmaelAmarochi, 2022-02-04:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [x] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

9. Please describe the issue in detail.

Is it possible to adapt the turtlebot_follower package to detect obstacles during the following of the person? The robot already reacts to the person and follows it, but what if something/someone was to interrupt the following. Is it possible to make the robot stop and or avoid the current obstacle/person and keep following the intended user? Thanks for your time.

ashekim, 2022-02-10:
Hi. @IsmaelAmarochi.

Turtlebot3_fallower is an example of using only lidar without using a camera.
In other words, the Turtlebot3_fallower recognize and determine objects only with the lidar scan value.
Turtlebot3_fallower recommends running in an environment there are no obstacles around.
If you want to distinguish between legs and obstacles, I recommend adding a camera.

Thank you.

IsmaelAmarochi, 2022-02-10:
Interesting, thank you for your response. It is highly appreciated.

How can I integrate the usage of a camera into the turtlebot_follower? Does the turtlebot_follower node perform the same on every turtlebot model? What happens if I use a Waffle Pi instead of a Burger model, would the node perform the same or would it make use of the camera. 

Is there a way to make the node use the camera? 

Could you enlighten me about this issue and could you provide any further tips on how to achieve this. The goal of my project is to make a human following robot. Is it possible to combine the follower with additional libraries in order to make the more aware of its surroundings and make it follow one person only while avoiding any obstacles?

Thank you.

ashekim, 2022-02-11:
@IsmaelAmarochi 
Hi again :)
First of all, please understand that it is difficult to support new features.

In the turtlebot3_folloer example, It is possible to change the model to Waffle Pi instead of Burger.
However, please note that leg recognition may not be smooth as the Lidar position changes.

Autorace is one example of using a camera with TB3. I am attaching a [link](https://emanual.robotis.com/docs/en/platform/turtlebot3/autonomous_driving/#autonomous-driving) .
Additionally, it would be appreciated if you refer to the example of [tracking people](https://youtu.be/t0IRNhu2Vjk).

Thank you.

closed 2024-11-13 by sunghowoo