ROBOTIS-GIT/turtlebot3 #828: not recieving published msgs from lidar sensor (LDS-02)

huybeme, 2022-02-15:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [x] etc (Please specify your OS here)
    sorry for my illiteracy but I have ubuntu focal 20.04.3 installed. I am not sure if this is the same as MATE
    

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
    couldn't find software or firmware version in bringup but I have attached a .txt file of it

7. Specify the commands or instructions to reproduce the issue.
      ros2 topic echo /scan
    

8. Copy and Paste the error messages on terminal.

    No error message

9. Please describe the issue in detail.

    I am simply trying to either echo my LDS-02 using the ld08_driver through the /scan topic. I have also tried to create a subscriber node to get the published msgs, please refer to the lidar_data.txt.
    My turtlebot is fairly new and I am hoping that it is not a faulty sensor. The sensor does power up and spin on power up.
 
[bringup_msg.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/8064974/bringup_msg.txt)
[lidar_data.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/8064984/lidar_data.txt)

ROBOTIS-Will, 2022-02-15:
Hi @huybeme 

The LDS-02 updated TB3 was released but I don't think it has not been shipped out oversea yet.
The driver works for a specific sensor of TurtleBot3.
Thanks.

huybeme, 2022-02-15:
Hi @ROBOTIS-Will 

Thank you for your response and I can confirm your suggestion that I do have LDS-01. The echo is returning data.

Thanks again!

closed 2022-02-15 by huybeme