ROBOTIS-GIT/turtlebot3 #829: catkin_make failed - turtlebot3

Somto-Dera, 2022-02-15:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ x ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ x ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ x ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ x ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ x ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - $ cd ~/catkin_ws/src
    - $ git clone -b develop https://github.com/ROBOTIS-GIT/ld08_driver.git
    - $ cd ~/catkin_ws/src/turtlebot3 && git pull
    - $ cd ~/catkin_ws && catkin_make

8. Copy and Paste the error messages on terminal.

    - HERE
    - -- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'turtlebot3'
-- ==> add_subdirectory(turtlebot3/turtlebot3)
-- +++ processing catkin package: 'turtlebot3_description'
-- ==> add_subdirectory(turtlebot3/turtlebot3_description)
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkin_workspace.cmake:121 (add_subdirectory):
  The source directory

    /home/ubuntu/catkin_ws/src/turtlebot3/turtlebot3_description

  does not contain a CMakeLists.txt file.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- +++ processing catkin package: 'turtlebot3_navigation'
-- ==> add_subdirectory(turtlebot3/turtlebot3_navigation)
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkin_workspace.cmake:121 (add_subdirectory):
  The source directory

    /home/ubuntu/catkin_ws/src/turtlebot3/turtlebot3_navigation

  does not contain a CMakeLists.txt file.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- +++ processing catkin package: 'ld08_driver'
-- ==> add_subdirectory(ld08_driver)
-- +++ processing catkin package: 'turtlebot3_bringup'
-- ==> add_subdirectory(turtlebot3/turtlebot3_bringup)
-- +++ processing catkin package: 'turtlebot3_example'
-- ==> add_subdirectory(turtlebot3/turtlebot3_example)
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkin_workspace.cmake:121 (add_subdirectory):
  The source directory

    /home/ubuntu/catkin_ws/src/turtlebot3/turtlebot3_example

  does not contain a CMakeLists.txt file.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- +++ processing catkin package: 'turtlebot3_slam'
-- ==> add_subdirectory(turtlebot3/turtlebot3_slam)
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkin_workspace.cmake:121 (add_subdirectory):
  The source directory

    /home/ubuntu/catkin_ws/src/turtlebot3/turtlebot3_slam

  does not contain a CMakeLists.txt file.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- +++ processing catkin package: 'turtlebot3_teleop'
-- ==> add_subdirectory(turtlebot3/turtlebot3_teleop)
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkin_workspace.cmake:121 (add_subdirectory):
  The source directory

    /home/ubuntu/catkin_ws/src/turtlebot3/turtlebot3_teleop

  does not contain a CMakeLists.txt file.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

9. Please describe the issue in detail.

    - Encountered this error while updating the drivers for the LDS driver(LDS-02) for the turtlebot3 burger. Faced the similar issues while updating the LDS drivers my turtlebot waffle pi that's running on Foxy. I'm a newb, any help provided will be appreciated.

ROBOTIS-Will, 2022-02-16:
Hi @Somto-Dera 

Thank you for reporting the issue.
During the `git pull` command, unnecessary updates are pulled into the `turtlebot3` directory.
Please run the command below to remove those unused directories.
Thank you.

```
$ cd ~catkin_ws/src/turtlebot3
$ rm -rf turtlebot3_slam/ turtlebot3_navigation/ turtlebot3_description/ turtlebot3_teleop/ turtlebot3_example/
$ cd ~catkin_ws && catkin_make
```

The eManual will be updated within a couple hours.
Thank you!

Somto-Dera, 2022-02-16:
Acknowledged. Thank you.

closed 2022-03-11 by darkUbermensch