ROBOTIS-GIT/turtlebot3 #834: TurtleBot3 movements are inverted in RVIZ

yipeyayeah, 2022-02-20:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - `roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=`
    - Estimate Initial Pose of robot
    - Set Navigation Goal via `2D Nav Goal`
    - Inverted movement of robot
8. Copy and Paste the error messages on terminal.

    - [image]

9. Please describe the issue in detail.

    - I am using ROS Noetic, and have a real physical TurtleBot3 with Lidar Sensor onboard. The teleop code to control the TurtleBot3 works fine during the map generation process, w moves the robot forward etc. However, when trying to perform point-to-point navigation `roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=` in RViZ using the `2D Nav Goal`, the TurtleBot3 movements are all inverted. Instead of moving forward to the navigation point, the TurtleBot3 actually moves backward. I have seen this question being asked in [here](https://answers.ros.org/question/32271/turtlebot-forwardbackward-movement-inverted-in-rviz/), and the answer by AjayKumar 1 is to invert the laser scan topic. However, I can't find that setting in RVIZ, must I manually configure some param files in the turtlebot3_navigation.launch file to solve this error?

ashekim, 2022-02-22:
@yipeyayeah 
Hello. Thank you for asking

After following the Local / Global route created by navigation, if it is determined that the route cannot be followed according to the set parameters, the recovery behavior is retryed after several times or the route is aborted in case of repeated failure.

When entering an area with potential collisions, even if you set a navigation goal point in front, Navigation create a path that moves backwards and move.

We recommend that you try to navigate by placing the TB3 at a certain distance from the wall.

thank you.

sunghowoo, 2024-11-13:
This issue will be closed due to prolonged inactivity. If the issue persists, feel free to reopen it.

closed 2024-11-13 by sunghowoo