ROBOTIS-GIT/turtlebot3 #842: Control lane function does not acknowledge updated parameters, only runs with defaults.

kylehannibal, 2022-03-10:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [X ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ X] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ X] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [X ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [X ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - roslaunch turtlebot3_gazebo turtlebot3_autorace_2020.launch
    - roslaunch turtlebot3_autorace_camera intrinsic_camera_calibration.launch
    - roslaunch turtlebot3_autorace_camera extrinsic_camera_calibration.launch
    - roslaunch turtlebot3_autorace_detect detect_lane.launch mode:=calibration
    - close previous command after calibration
    - edit lane.yaml file with new values
    - roslaunch turtlebot3_autorace_detect detect_lane.launch mode:=action
    - roslaunch turtlebot3_autorace_driving turtlebot3_autorace_control_lane.launch
    - (on the turtlebot) 
    - roslaunch turtlebot3_bringup turtlebot3_robot.launch

8. Copy and Paste the error messages on terminal.

    - No error messages, the robot just refuses to acknowledge updated parameters. If the turtlebot is forced to use the new parameters, it won't run. The only option is to run the turtlebot with the old parameters, where it doesn't detect lines. 

9. Please describe the issue in detail.

    - Read above message

ashekim, 2022-03-21:
Hi. @kylehannibal. 
Save the parameter value of the [turtlebot3_autorace_detect/param/lane/lane.yaml](https://github.com/ROBOTIS-GIT/turtlebot3_autorace_2020/blob/master/turtlebot3_autorace_detect/param/lane/lane.yaml) file and run it.
If it still doesn't work,
Change the parameter value in [detect_lane#line_33~45](https://github.com/ROBOTIS-GIT/turtlebot3_autorace_2020/blob/ab2a1f1520ee62eebca3cc6e534b297613fb5d75/turtlebot3_autorace_detect/nodes/detect_lane#L33), and run it after building.
Thank you.

sunghowoo, 2024-11-13:
This issue will be closed due to prolonged inactivity. If the issue persists, feel free to reopen it.