ROBOTIS-GIT/turtlebot3 #843: OS Problem

MalikaGawande, 2022-03-16:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?
2. 
    - [ ] Waffle Pi

3. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Melodic Morenia

4. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 4

5. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

6. Which OS you installed on Remote PC?

    
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)

7. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

8. Specify the commands or instructions to reproduce the issue.

    - HERE

9. Copy and Paste the error messages on terminal.

    - HERE

10. Please describe the issue in detail.

    - OS is not working which is provided on the official site. It is saying update and screen getting stuck. I'm having Raspberry Pi 4 B with 2GB RAM

ashekim, 2022-03-21:
Hi @MalikaGawande. 
It is recommended to install ubuntu 18.04 server instead of raspi based OS.

Please refer to the attached image and reinstall.
[image]

If it still doesn't work, please feel free to let us know.
Thank you.

MalikaGawande, 2022-03-21:
I tried using this image and I'm getting the following error which is shown
in image below and screen getting stuck on that display

ashekim, 2022-03-21:
@MalikaGawande 
I'd like to know where the problem is.
Have you proceeded to [3.2.6 Configure the WiFi Network Setting](https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#configure-the-wifi-network-setting)?
Thank you for attaching a picture where the screen freezes with an error.

ROBOTIS-Will, 2022-03-21:
Hi @MalikaGawande 

Could you upload the image via GitHub thread?
https://github.com/ROBOTIS-GIT/turtlebot3/issues/843
Replying to the email doesn't properly attach any images at all.
Thanks!

JKinsly, 2022-11-26:
Hi! I am experiencing the same problem. I was wondering if this issue has been solved? And if so what the solution is. I am using the same software and hardware.

closed 2024-11-13 by sunghowoo