ROBOTIS-GIT/turtlebot3 #845: Turtlebot3 Navigation2 parameter file not working

StijnGerbranda, 2022-03-28:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [X] Ubuntu preinstalled server (20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [X] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    This problem occurs on the turtle bot and in the simulation using Gazebo
    TurtleBot:
    - On the bot I run ''ros2 launch turtlebot3_bringup robot.launch.py use_sim_time:=False'' 
    - after adding these lines to the .bashrc File 
        - source /opt/ros/foxy/setup.bash
        - export TURTLEBOT3_MODEL=burger

    - On a virtual machine running Ubuntu 20.04 after confirming a correct connection.
    - I run "ros2 launch turtlebot3_navigation2 navigation2.launch.py map:=/home/turtlebot/map3.yaml use_sim_time:=False" Selecting the map of the environment the bot is reciting.

  Simulation:
   - In different terminals we run:
       - export TURTLEBOT3_MODEL=burger
       - ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
   - and
       - export TURTLEBOT3_MODEL=burger
       - ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=True map:=/home/turtlebot/map3.yaml (again selecting the map of my environment.)
   - This starts the simulation and shows correct results

8. Copy and Paste the error messages on terminal.

    - There is no error message. The problem is changing the parameters of the navigation package.

9. Please describe the issue in detail.

    - When changing the parameters in the navigation file or directly using the console nothing changes.
    - According to https://emanual.robotis.com/docs/en/platform/turtlebot3/navigation/#tuning-guide
    Changing the values in turtlebot3_navigation2/param/${TB3_MODEL}.yaml (turtlebot3_navigation2/param/burger.yaml, in my case) Should allow to edit the navigation settings. However changing this file does not appear to work and the changes values don't appear anywhere else. 
   - Changing the value manually after starting the program using "ros2 param set" seem to be changing the value correctly but don't change the behavior or visualization. 
   - Changing other files that seem to have the same variables does not change the result.
   - I tried this but without result: "https://github.com/ros-planning/navigation2/issues/2651"

The burger.yaml file and the variations I tried: (line 104)
[Original_burger.yaml.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/8362323/Original_burger.yaml.txt)
[Variation1_burger.yaml.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/8362321/Variation1_burger.yaml.txt)
[Variation2_burger.yaml.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/8362322/Variation2_burger.yaml.txt)

   
So in conclusion how do I change the parameters of the navigation package or how do I get the param file to work

ashekim, 2022-03-30:
Hi @StijnGerbranda.
Thank you for your questions.

When I checked the attached file, it seems to be different from the .yaml format we [provided](https://github.com/ROBOTIS-GIT/turtlebot3/blob/foxy-devel/turtlebot3_navigation2/param/burger.yaml).
Please check the attached [link](https://github.com/ROBOTIS-GIT/turtlebot3/blob/8237b796ea1571033bf3230fbc78d1143968ddd1/turtlebot3_navigation2/param/burger.yaml#L154) and correct the parameter values of **lines 154 and 155.**

I am attaching an image of the result of my execution below.

---
inflation_radius: 1.0
cost_scaling_factor: 3.0

[image]

---
inflation_radius: 0.2
cost_scaling_factor: 20.0

[image]

Please test it and if it doesn't work, feel free to let us know.
Thank you.

StijnGerbranda, 2022-03-31:
Thank you, @ashekim

This did not immediately solve the issue but it helped. 
After finding these pages:
https://navigation.ros.org/tuning/index.html and  https://answers.ros.org/question/371814/print-agruments-in-ros-2-foxy-launch-files/

I was able to repair the navigation2.launch.py file and manually select a copy of nav2_params.yaml in the package’s params/ directory that I could edit. By doing this I could change the parameters but using my original burger.yaml file crashed the program. using the new burger.yaml file seems to fix that issue.

closed 2022-03-31 by StijnGerbranda