ROBOTIS-GIT/turtlebot3 #846: rviz "no transform" error after launching turtlebot3 burger slam simulation

dtwind62, 2022-03-29:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [X ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ X] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ X] Ubuntu preinstalled server (18.04/20.04) 20.04
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [X ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x] simulation, not using robot
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.
roscore
export TURTLEBOT3_MODEL=burger
roslaunch turtlebot3_gazebo turtlebot3_world.launch

8. Copy and Paste the error messages on terminal.
ROS_HOSTNAME: 192.168.1.34
ROS_IP: 192.168.1.34
ROS_MASTER_URI: http://192.168.1.34:11311
TURTLEBOT3_MODEL: burger
don@rosnoetic:~$ roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping
... logging to /home/don/.ros/log/ae092e4c-aefe-11ec-9a2d-61aadcf943e5/roslaunch-rosnoetic-14673.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://192.168.1.34:46697/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 50.0
 * /robot_state_publisher/tf_prefix: 
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /turtlebot3_slam_gmapping/angularUpdate: 0.2
 * /turtlebot3_slam_gmapping/astep: 0.05
 * /turtlebot3_slam_gmapping/base_frame: base_footprint
 * /turtlebot3_slam_gmapping/delta: 0.05
 * /turtlebot3_slam_gmapping/iterations: 5
 * /turtlebot3_slam_gmapping/kernelSize: 1
 * /turtlebot3_slam_gmapping/lasamplerange: 0.005
 * /turtlebot3_slam_gmapping/lasamplestep: 0.005
 * /turtlebot3_slam_gmapping/linearUpdate: 1.0
 * /turtlebot3_slam_gmapping/llsamplerange: 0.01
 * /turtlebot3_slam_gmapping/llsamplestep: 0.01
 * /turtlebot3_slam_gmapping/lsigma: 0.075
 * /turtlebot3_slam_gmapping/lskip: 0
 * /turtlebot3_slam_gmapping/lstep: 0.05
 * /turtlebot3_slam_gmapping/map_frame: map
 * /turtlebot3_slam_gmapping/map_update_interval: 2.0
 * /turtlebot3_slam_gmapping/maxUrange: 3.0
 * /turtlebot3_slam_gmapping/minimumScore: 50
 * /turtlebot3_slam_gmapping/odom_frame: odom
 * /turtlebot3_slam_gmapping/ogain: 3.0
 * /turtlebot3_slam_gmapping/particles: 100
 * /turtlebot3_slam_gmapping/resampleThreshold: 0.5
 * /turtlebot3_slam_gmapping/sigma: 0.05
 * /turtlebot3_slam_gmapping/srr: 0.1
 * /turtlebot3_slam_gmapping/srt: 0.2
 * /turtlebot3_slam_gmapping/str: 0.1
 * /turtlebot3_slam_gmapping/stt: 0.2
 * /turtlebot3_slam_gmapping/temporalUpdate: 0.5
 * /turtlebot3_slam_gmapping/xmax: 10.0
 * /turtlebot3_slam_gmapping/xmin: -10.0
 * /turtlebot3_slam_gmapping/ymax: 10.0
 * /turtlebot3_slam_gmapping/ymin: -10.0

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    turtlebot3_slam_gmapping (gmapping/slam_gmapping)

auto-starting new master
process[master]: started with pid [14688]
ROS_MASTER_URI=http://192.168.1.34:11311

setting /run_id to ae092e4c-aefe-11ec-9a2d-61aadcf943e5
process[rosout-1]: started with pid [14698]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [14705]
process[turtlebot3_slam_gmapping-3]: started with pid [14706]
process[rviz-4]: started with pid [14707]

[image]
9. Please describe the issue in detail.
after launching the slam mapping simulation, I get a ROBOTMODEL error, it says "no transformation/ This is a new install, I believe I have followed all the instructions in the manual. I get this same error when I try to use the turtlebot3 burger robot.

ashekim, 2022-03-29:
Hi @dtwind62.
Thank you for your questions.

I want to know if you entered `export TURTLEBOT3_MODEL` before launching slam.
Thank you.

dtwind62, 2022-03-29:
Thanks for the reply, I actually have export TURTLEBOT3_MODEL=burger as the
last entry in my .basic file.

dtwind62, 2022-03-29:
Oops .bashrc file

ashekim, 2022-03-30:
Hello @dtwind62.
Thank you for your reply.

I checked the attached image again.
It seems that Gazebo did not launch normally.
I found that Rviz throws a No transform error when Gazebo is not running.

Could you please check if Gazebo is running normally?
Thank you.

ashekim, 2022-03-30:
Additionally, I am attaching an image that I tested

* When GAZEBO is executed
[image]

* When GAZEBO is not executed
[image]

Thank you.

dtwind62, 2022-03-30:
first I launched {roscore}, then I launched {roslaunch turtlebot3_gazebo turtlebot3_world.launch} and gazebo launched with no error, and as soon as I launched {roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping}, I got an illegal instruction (core dump) in my gazebo window.
[image]
[image]
this is the error that popped up in my gazebo terminal :
Illegal instruction (core dumped)
[gazebo-1] process has died [pid 1746, exit code 132, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode /home/don/catkin_ws/src/turtlebot3_simulations/turtlebot3_gazebo/worlds/turtlebot3_world.world __name:=gazebo __log:=/home/don/.ros/log/a2689984-b056-11ec-a73e-4dd067d47d9c/gazebo-1.log].
log file: /home/don/.ros/log/a2689984-b056-11ec-a73e-4dd067d47d9c/gazebo-1*.log

ashekim, 2022-04-01:
Hi @dtwind62.
There seems to be no problem with the setup and execution process....

There are cases where errors occur with various external variables in ROS, so it is recommended to run it after reinstalling the ROS package. 

If you still have the same problem after trying, please feel free to contact us.
Thank you.

dtwind62, 2022-04-01:
I just noticed, in my slam terminal window I'm getting this warning.

Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom at time 60.003000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp

ashekim, 2022-04-06:
Hello @dtwind62.
Thanks for sharing the warning content.
Checking the issue, it seems to be related to ros/geometry2#467.

After we figure out the problem, we will fix it.
It may take more time.
Thank you for your patience.

johnreb20, 2023-04-18:
Hi @ROBOTIS-Ashe,

I am also getting the rviz robot model error. I reinstalled it again but no progress.

Thanks,
John.

tharindurm, 2023-04-22:
Hi @johnreb20 
It seems the robot_state_publisher is not running. Hence Rviz cannot figure out which is where. You can check this by running `rosnode list` command. 'robot_state_publisher' name should be in the list if its running.

1. First run any turtlebot3 world 
2. Then copy the below code in to a separate launch file or merge the content with one of your own launch file and execute it. This should run the robot_state_publisher node
`<launch>`
 `      <node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher">`
 `         <param name="publish_frequency" type="double" value="50.0" />`
 `         <param name="tf_prefix" value="/"/>`
 `      </node>`
`</launch>`
3. Run rviz and check

Hope this helps.

closed 2024-11-13 by sunghowoo