ROBOTIS-GIT/turtlebot3 #848: Robot_launch.py  process has died [pid 6524, exit code -11

muratcngncr, 2022-03-29:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [x] etc (ROS 2 Galactic Geochelone)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [x] etc (nvidia jetson nano)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [x] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - ros2 launch turtlebot3_bringup robot.launch.py 

8. Copy and Paste the error messages on terminal.

    - [INFO] [launch]: All log files can be found below /home/turtle/.ros/log/2022-03-29-16-47-01-147040-nano-6518
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
[INFO] [robot_state_publisher-1]: process started with pid [6520]
[INFO] [hlds_laser_publisher-2]: process started with pid [6522]
[INFO] [turtlebot3_ros-3]: process started with pid [6524]
[hlds_laser_publisher-2] [INFO] [1648561622.295262462] [hlds_laser_publisher]: Init hlds_laser_publisher Node Main
[hlds_laser_publisher-2] [INFO] [1648561622.295478921] [hlds_laser_publisher]: port : /dev/ttyUSB0 frame_id : base_scan
[turtlebot3_ros-3] [INFO] [1648561622.298224546] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [1648561622.299198244] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-3] [INFO] [1648561622.299656369] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [INFO] [1648561622.300054962] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
[robot_state_publisher-1] Link base_link had 5 children
[robot_state_publisher-1] Link caster_back_link had 0 children
[robot_state_publisher-1] Link imu_link had 0 children
[robot_state_publisher-1] Link base_scan had 0 children
[robot_state_publisher-1] Link wheel_left_link had 0 children
[robot_state_publisher-1] Link wheel_right_link had 0 children
[... 57 lines cut ...]
[turtlebot3_ros-3] [ERROR] [1648561670.801702627] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561671.951467834] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561673.issue051711111] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561673.851588217] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561674.702311482] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561677.602197210] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561677.701821892] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561679.251843052] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561680.001746121] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561684.101650246] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561685.301905293] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561690.101822838] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561691.501759228] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561692.052408905] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561692.352129824] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561693.502350854] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561694.752273153] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561696.952526172] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561697.803289349] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[turtlebot3_ros-3] [ERROR] [1648561700.754069682] [DynamixelSDKWrapper]: Failed to read[[TxRxResult] Incorrect status packet!]
[ERROR] [turtlebot3_ros-3]: process has died [pid 6524, exit code -11, cmd '/opt/ros/galactic/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 --ros-args --params-file /opt/ros/galactic/share/turtlebot3_bringup/param/burger.yaml'].

9. Please describe the issue in detail.

    - Hi, i just started to work with ROS2 , when i launch the robot.launch.py file starts to giving DynamixelSDKWrapper Failed to read [TxRxResult] Incorrect status packet!] error and after while turtlebot3_ros node died. Until the turtlebot3_ros node died i can able to move my turtlebot and also i can take all the sensors data.

ashekim, 2022-03-30:
Hi @muratcngncr.
Thank you for your questions.

Unfortunately, Please note that the nvidia jetson nano + ROS 2 Galactic version has not been tested as it is not an officially supported platform.

Please check whether Dynamixel works through [Dynamixel Wizard 2.0](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/).
After uploading "usb to dxl" to OpenCR1.0, connect Dynamixel Wizard 2.0.
After connecting Dynamixel Wizard 2.0, please check using functions such as firmware recovery and self-diagnosis.

Thank you.

AbisoyeAkinloye, 2022-09-10:
**Unable to launch turtlebot3_bringup**
[x] Ubuntu 20.04 LTS (Focal Fossa)
[x] ROS 2 Foxy Fitzroy

`ros2 launch turtlebot3_bringup robot.launch.py`

**ERROR:**

[INFO] [launch]: All log files can be found below /home/loye/.ros/log/2022-09-10-13-16-05-502716-loye-18011
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=InvalidLaunchFileError('py')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 53, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
    return getattr(launch_file_module, 'generate_launch_description')()
  File "/home/loye/turtlebot3_ws/install/turtlebot3_bringup/share/turtlebot3_bringup/launch/robot.launch.py", line 33, in generate_launch_description
    LDS_MODEL = os.environ['LDS_MODEL']
  File "/usr/lib/python3.8/os.py", line 675, in __getitem__
    raise KeyError(key) from None
KeyError: 'LDS_MODEL'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 130, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [py]: 'LDS_MODEL'

closed 2024-11-13 by sunghowoo