ROBOTIS-GIT/turtlebot3 #849: How to fetch velocity value for joint_states

tonaaskarwur, 2022-04-04:
I am trying to publish my own joint_states for my own real robot, and I'm using turtlebot3 as my reference. My robot is running on ROS2 foxy Ubuntu 20.04. While trying to get my own joint_states, I need to convert my encoder ticks to radians for the joint_states' position, and velocity from RPM to MS.
For the later msg, my understanding is that we can get this value from linear and angular velocity.
In turtlebot3 repo on foxy branch, it is noted that the value of [RPM to MS = 0.229 * 0.0034557519189487725](https://github.com/ROBOTIS-GIT/turtlebot3/blob/foxy-devel/turtlebot3_node/include/turtlebot3_node/sensors/joint_state.hpp).
I am not sure how the maintainers can get the value noted above. Can anyone help to explain?
I opened the [reference](https://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#goal-velocity104) noted in the header file which directed me to turtlebot3's Profile acceleration, but still I can't get a good understanding how these values come up from that.
Thank you

ashekim, 2022-04-04:
Hi @tonaaskarwur.
Thank you for your questions.
Below is the process of unit conversion from RPM to m/s.

**RPM_to_MS = (2πr * RPM) / 60** 
(r = radius)

The Turtlebot3 wheel diameter is 66mm (=0.066m).
The RPM of the DXL used in the Turtlebot3 is 0.229.

Therefore, it can be calculated as RPM_to_MS = RPM * (2πr / 60),
Here you can see that 2πr / 60 = (2π * 0.033) / 60 = 0.00345575191.

Thank you.

closed 2022-04-04 by tonaaskarwur

foxydevelby, 2023-01-18:
how about the Tick?
**// 0.087890625[deg] * 3.14159265359 / 180 = 0.001533981f**
from where is the value 0.087890625[deg] derived or get from?