ROBOTIS-GIT/turtlebot3 #850: tutlebot_gazebo building error

nikboy111, 2022-04-06:
OS: Ubuntu 20.04 Xfce
ROS-Version: ROS2 Foxy

Hello,
when trying to set up the turtlebot3 simulation following the [e-manual](https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/) I keep running into the same problem over and over again:

> --- stderr: turtlebot3_gazebo                                             
CMake Error in CMakeLists.txt:
  Imported target "dart" includes non-existent path
    "/include"
  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
   The path was deleted, renamed, or moved to another location.
   An install or uninstall procedure did not complete successfully.
   The installation package was faulty and references files it does not
  provide.

Installing the turtlebot3_simulations package as a source file also didn't work for me.
Does anyone have an idea how to fix this issue? 
Thank you.
Sincerly Nik

ashekim, 2022-04-07:
Hi @nikboy111.
Thank you for your questions.
Your installation package seems to be faulty
Please refer to PC Setup in the [Quick Start Guide](https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#quick-start-guide-4) and reinstall.

Thank you.

nikboy111, 2022-04-07:
Hi @ashekim, 
thank you for your answer. I resinstalled ROS as well as all the packages mentioned in the Guide and still get the same error.

ashekim, 2022-04-08:
Hello. @nikboy111.
The problem seems to be caused by the dart package.
I am attaching an issue with symptoms similar to yours.

https://github.com/dartsim/dart/issues/1428

thank you.

closed 2022-04-11 by nikboy111

p79281, 2022-10-05:
Same issue as @nikboy111 here on Ubuntu 20.04. Any help welcome

Julien-Gustin, 2023-03-03:
Hi @shrinivas96 @p79281 @nikboy111 , does anyone of you managed to solve the problem?