ROBOTIS-GIT/turtlebot3 #851: the dynamicxel problem .

himhan34, 2022-04-08:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?
    - [ ] Burger
  

2. Which ROS is working with TurtleBot3?
    - [ ] ROS 1 Melodic Morenia

3. Which SBC(Single Board Computer) is working on TurtleBot3?
    - [ ] Raspberry Pi 3B+

4. Which OS you installed on SBC?
    - [ ] Ubuntu preinstalled server (18.04)

5. Which OS you installed on Remote PC?
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
   
6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version:  ros melodic 
    - Firmware version: opencr 0.1.0

7. Specify the commands or instructions to reproduce the issue.
i am newby to the burger, so i am trying to move burger in the keyboard which is in the tutorials. 

[image]

this is a thing that i tried. 

    
8. Copy and Paste the error messages on terminal.

   there is no error in the picture but my turtlebot3 is not moving while i tab the wadsx buttons on my keyboard

9. Please describe the issue in detail.
to solve them, i tried to reboot the software on the opencr, but  i got the message from  arduone serial monitor.

[image]

could you help me to solve it?

ROBOTIS-Shibata, 2022-04-09:
Wow,
Perhaps, you are running the wrong Arduino sketch.
The message is a sketch for rewriting Dynamixel's internal settings.
https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#setup-dynamixels-for-turtlebot3

You shall try OpenCR setup again.
https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup-1

himhan34, 2022-04-09:
thank you for replying my question. i 'll do it in monday!
after doing the procedure that you recommend i will reply that what happened, thanks for answering.

himhan34, 2022-04-14:
i tried to re-upload the opencr firmware, and all this went welll in the arduino. i saq the jupt _to_fw  in the arduino
but after doing the thing in the tutorial, i cant move the turtle bot3 in keyboard. 

i also checked the dynamixel with the dynamixel wizard 2, and 2 dynamixels were rotating when i checked through computer. 
but i cant move in the ros system. how can i do?

shiba-8ro, 2022-04-14:
Hi,
If two Dynamixels have the same ID, it will not work properly.
Are the two Dynamixels different ID? The left side of the robot is ID1 and the right side is ID2. In addition, Baudrate is 1Mbps. Dynamixel with ID2 needs to be set in reverse.
I'm worried that you may have run a program from OpenCR that manipulates Dynamixel's internal settings. If you can use Dynamixel Wizard 2.0, check out the details of Dynamixel's internal settings.

himhan34, 2022-04-14:
can you explain the meaning of "Dynamixel with ID2 needs to be set in reverse" ?  
i think i don't change the seting with the id. i'll try that either.

shiba-8ro, 2022-04-14:
It is setting of Drive mode(10) in Dynamixel control table.
https://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#drive-mode10
[image]

himhan34, 2022-04-14:
how can i change the mode?

himhan34, 2022-04-15:
[image]

i  also have a question about the error message. 
the socket is already in use in this photo. how can i solve it?

ROBOTIS-Will, 2022-04-19:
@himhan34 
Looks like your network setup is not correctly done.
Please make sure to set up the network configuration on both your PC and TurtleBot3.
Both your PC and TurtleBot3(Raspbery PI) should have an identical ROS_MASTER_URI, while ROS_HOSTNAME should be their own IP addresses.
[image]

himhan34, 2022-04-22:
thank you for answering my question, i also chaeck the ip address, and both was same, 
i resolve the question that i had, but i have another problem 

i am following the https://emanual.robotis.com/docs/en/platform/turtlebot3/basic_operation/#basic-operation  instruction above, but i cant move the turtlebot, do you know why, i also checked that, i also uploaded the firmware of opencr. 
 
but when i press the wdax buttun in my keyboard  my turtlebot is not moving.  

do you know why?

closed 2022-06-14 by himhan34