ROBOTIS-GIT/turtlebot3 #853: turtlebot3 , opencr ssh

ohyousung, 2022-04-10:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [v] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [v ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)
  
3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [v ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [v ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ v] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [v ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - If Raspberry Pi and opencr are connected to each other, ssh remote access is not possible. I can't connect to the iptime router remotely.

If Raspberry Pi , opencr are separated from each other, remote access is possible.

bashrc file ip I checked that they match each other.

9. Please describe the issue in detail.

    - 

[image]

ROBOTIS-Will, 2022-04-12:
Hi @ohyousung 
Could you check if your RPi is successfully and fully booted up when it is connected to the OpenCR?
Thank you.

shiba-8ro, 2022-04-12:
Hi @ohyousung 
Be careful. The cable connecting that supplies DC5V from OpenCR to Raspi is incorrect.
[image]

This is collect ↓
[image]

sunghowoo, 2024-11-13:
This issue will be closed due to prolonged inactivity.

closed 2024-11-13 by sunghowoo