ROBOTIS-GIT/turtlebot3 #855: ld08 driver, how to test LDS-02?

Jeonghyeongseoc, 2022-05-02:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [o ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [o ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [o ] etc (latte panda)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [o ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [o ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [ROS]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

$ sudo apt update
$ sudo apt install libudev-dev
$ cd ~/catkin_ws/src
$ git clone -b develop https://github.com/ROBOTIS-GIT/ld08_driver.git
$ cd ~/catkin_ws/src/turtlebot3 && git pull
$ rm -r turtlebot3_description/ turtlebot3_teleop/ turtlebot3_navigation/ turtlebot3_slam/ turtlebot3_example/
$ cd ~/catkin_ws && catkin_make

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.
   I installed ld08 driver but LDS-02 is not work. Plase tell me how to test LDS-02.
There is a mention about LDS-01 in 'emanual.robotis.com' , but no LDS-02 how to test. 

  And  " cd ~/catkin_ws/src/turtlebot3 && git pull"  line is not work.

Jeonghyeongseoc, 2022-05-02:
@Will son

ROBOTIS-Will, 2022-05-03:
Hi @Jeonghyeongseoc 

Make sure that your LDS model is the LDS-02
[image]
The LD08 driver is not compatible with the previous LDS-01 model.
Depending on your method of TB3 package installation, pulling the source code may not necessary.

For example, if you have installed the `turtlebot3` package with `sudo apt install ros-melodic-turtlebot3` command(debian package install), you won't have the ~/catkin_ws/src/turtlebot3 directory, therefore, the command will fail.

You may try uninstalling the debian package and download the source code then build from it as below.
Below commands should be applied on your SBC, not on the PC.

```bash
$ sudo apt remove ros-melodic-turtlebot3
$ cd ~/catkin_ws/src
$ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/turtlebot3.git
$ cd ~/catkin_ws/src/turtlebot3
$ rm -r turtlebot3_description/ turtlebot3_teleop/ turtlebot3_navigation/ turtlebot3_slam/ turtlebot3_example/
$ cd ~/catkin_ws && catkin_make
```

At the moment, we do not provide a separate tool to test the sensor since every LDS-02 is tested before the packaging.
I'll try to create a test example in the future.

Thank you.

SunilSrivatsav9, 2023-08-15:
LDS-2  ( ld-08 driver is available for ros2 (https://github.com/ROBOTIS-GIT/ld08_driver) 
WHere can I find driver for ROS1?

shiba-8ro, 2023-08-21:
@SunilSrivatsav9 
Can the below URL link help you?
https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#new-lds-02-configuration-2

SunilSrivatsav9, 2023-08-21:
It helped me Thank you

sunghowoo, 2024-11-13:
This issue appears to be resolved, so it will be closed.

closed 2024-11-13 by sunghowoo