ROBOTIS-GIT/turtlebot3 #857: turtlebot3 Burger wifi not connecting, Noetic.

mayBeAutomata, 2022-05-11:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ x ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ x ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ x ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ x ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ x ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - I followed the SBC Noetic setup on the emanual.

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    - rpi doesn't connect to wifi. I use the same wifi for my turtlebot3 waffle pi (rpi4, ROS2 Foxy), and it works fine. I have made the regular adjustments to  "50-cloud-init.yaml" file. 
    - 
    - Imager =>           "
[image]
imager_1.7.2_amd64.deb"

ROBOTIS-Will, 2022-05-16:
@mayBeAutomata 
In case the Raspberry Pi and your wifi router uses different WiFi channel, this could happen.
Please try setting your country code in the `/etc/default/crda` file as below.
By default the REGDOMAIN is empty.
```
REGDOMAIN=US
```

mayBeAutomata, 2022-05-16:
Noted. Will try that and update you ASAP.

mayBeAutomata, 2022-06-03:
Issue Resolved.

closed 2022-06-03 by mayBeAutomata