ROBOTIS-GIT/turtlebot3 #858: Problem with Navigation Turtlebot3

BENBEY-I, 2022-05-12:
ISSUE TEMPLATE ver. 0.4.0

Hi everyone, 

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [O] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [O] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [O] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [O] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [O] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - turtlebot3 navigation

8. Copy and Paste the error messages on terminal.

    [ WARN] [1652349909.834607074]: Costmap2DROS transform timeout. Current time: 1652349909.8345, global_pose stamp: 1652349905.8794, tolerance: 0.5000
[ERROR] [1652349910.081358083]: Rotate recovery can't rotate in place because there is a potential collision. Cost: -1.00
[ WARN] [1652349910.081689770]: Transform timeout for global_costmap. Current time: 1652349910.0816, pose stamp: 1652349906.1826, tolerance: 0.5000
[ WARN] [1652349910.082083003]: Unable to get starting pose of robot, unable to create global plan
[ WARN] [1652349910.183285280]: Unable to get starting pose of robot, unable to create global plan
[ INFO] [1652349910.381806723]: Got new plan
[ERROR] [1652349910.382052721]: Could not get robot pose
[ERROR] [1652349910.382149315]: Could not get robot pose
[ WARN] [1652349910.454463224]: Could not get robot pose, cancelling reconfiguration
[ WARN] [1652349910.481901106]: Clearing both costmaps to unstuck robot (3.00m).
[ERROR] [1652349910.482295824]: Cannot clear map because pose cannot be retrieved
[ERROR] [1652349910.482513397]: Cannot clear map because pose cannot be retrieved
[ WARN] [1652349910.482942601]: Unable to get starting pose of robot, unable to create global plan

9. Please describe the issue in detail.

Problem 1 : 

It take a long time to refresh the image in RViz, is the problem from the robot take time to send the updates to my PC or it's the PC running slowly ? is there any frequency i can change to speed up the communication between them? 

Problem 2 (main problem) : 

in the first time i had the time sync problem now it's sync, 

[image]

But know when i use the 2D Nav Goal and after setting the robot in his place with 2D Pose Estimate and moving the robot right and left to collect the data for the AMCL, the Robot starts turning around himself, and sometimes the model robot in RViz start changing his place while the real robot is turning in his place, and sometimes the robot start to move toward the goal then he just starts turning around himself again.

Problem 3 : 

The robot detect sometimes an obstacle doesn't exist in the reality 

Problem 4 :

When the robot turn to the left or right the obstacles get staked in the robot's system and get in the way of the robot so he can't calculate the road 

[image]

Question 1 :

How can i solve these problems especially the Problem 2 ? The navigation doesn't work !! 

Question 2 : 

in my last test 

[image]

there is a red line and a black line going to the goal, red line for the global_map, and the black for the local_map, what is the difference ? and he should use the Local or Global ?

ROBOTIS-Will, 2022-05-13:
@BENBEY-I 
Hi, when performing the navigation, please try running it in a space where enough space is allowed.
Since navigation in a narrow space requires detailed tuning, it is not recommended to run from the beginning.

When setting the initial pose of the robot, please use the teleop_key instead of sending the robot with `2D Nav Goal` button. This is because the robot may not be able to calculate a proper path under certain environment.

Please see the difference of the Local and Global planner in the links below.
http://wiki.ros.org/global_planner
http://wiki.ros.org/base_local_planner

Basically robot uses the global path to get to the goal, while using the local path to avoid obstacles.

Thank you.

ROBOTIS-Will, 2022-05-16:
@BENBEY-I 
You mean the frequency of publisher?
You can increase the `loop_rate`, but you may not increase the frequency too fast due to the bottleneck between various devices so at some point, you may not get a desired frequency.
Usually the robot publishes topics and the PC subscribes sensor data and visualizes on the screen.
https://stackoverflow.com/questions/56565275/can-you-explain-this-syntax-rosrate-loop-rate10