ROBOTIS-GIT/turtlebot3 #861: Incorrect reading from LDS-01

elimas9, 2022-05-18:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [X] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [X] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [X] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [X] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [0.0.0]
    - Firmware version: [1.2.3]

7. Specify the commands or instructions to reproduce the issue.

Reading the /scan topic on ROS, but also directly from the LDS-01 laser scanner connected to a PC with the code [hls_lfcd_lds_driver](https://github.com/ROBOTIS-GIT/hls_lfcd_lds_driver)
    
8. Copy and Paste the error messages on terminal.

 Most of the time the second half of the laser scanner range or intensities vector is filled with zeros (corrupted data). The range of corrupted data received from the sensor increases if the laser scanner has been recently on for a long time.

9. Please describe the issue in detail.

We have an issue with the LDS-01 laser scanner, in which we receive numerous zeros in the range reading. I tried checking with the code in [hls_lfcd_lds_driver](https://github.com/ROBOTIS-GIT/hls_lfcd_lds_driver), and it seems that most of the time the second half of the range vector is filled with zeros (corrupted data). The range of corrupted data received from the sensor increases if the laser scanner has been recently on for a long time.
  
  What can be the cause of this?

ROBOTIS-Will, 2022-05-19:
Hi @elimas9 

It seems like the issue is caused by the LDS-01.
If the direct reading from the sensor via USB2LDS interface is missing half of the data, the sensor could be malfunctioning.
I'd recommend to get the sensor examined and repaired / replaced if necessary.
Thank you.

elimas9, 2022-05-23:
Hi,
thanks a lot for your quick reply!
Since it is the second LDS-01 with which we have some hardware issues, do you suggest to try with this other model RPLIDAR A1 ( https://www.slamtec.com/en/Lidar/A1 )?
It seems compatible with the Turtlebot3 and ROS.
Thank you!

closed 2022-05-23 by elimas9

elimas9, 2022-05-24:
Great, thank you :)