ROBOTIS-GIT/turtlebot3 #863: Turtlebot3 Waffle_Pi - Raspberry Pi Camera Not working/Disabled

Hereisemi, 2022-05-19:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

Not sure where I can find it: 
[image]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    Everything is working expect the Raspberry Pi Camera:

[image]

Apparently the Camera is not enabled, but i dont have any option to activate it on my Turtlebot

[image]

I already tried to "sudo apt-get update" and "sudo apt-get upgrade" and i tried to update it in the "sudo raspi-config". It says "its already the newest version"

[image]

Any idea how i can proceed? 

Thank you!

ROBOTIS-Will, 2022-05-25:
Hi @Hereisemi 
I'm installing opencv and testing the RPi Cam on Foxy.
I'll get back to you once it's done.
Thank you!

Hereisemi, 2022-05-26:
> 

Thank you! Im looking forward to hear from you :)

ROBOTIS-Will, 2022-05-30:
**Updates**
Building the opencv on RPi3B+ (the same steps had been applied to Ubuntu20.04 + Noetic for TB3 Autorace).
This takes several hours to complete.
Below is the list of commands that I used (at the moment, I'm building source with `make -j4` command)

```bash
$ sudo apt-get install build-essential cmake gcc g++ git unzip pkg-config
$ sudo apt-get install libjpeg-dev libpng-dev libtiff-dev libavcodec-dev libavformat-dev libswscale-dev libgtk2.0-dev libcanberra-gtk* libxvidcore-dev libx264-dev python3-dev python3-numpy python3-pip libtbb2 libtbb-dev libdc1394-22-dev libv4l-dev v4l-utils libopenblas-dev libatlas-base-dev libblas-dev liblapack-dev gfortran libhdf5-dev libprotobuf-dev libgoogle-glog-dev libgflags-dev protobuf-compiler
$ cd ~
$ wget -O opencv.zip https://github.com/opencv/opencv/archive/4.5.0.zip
$ wget -O opencv_contrib.zip https://github.com/opencv/opencv_contrib/archive/4.5.0.zip
$ unzip opencv.zip
$ unzip opencv_contrib.zip
$ mv opencv-4.5.0 opencv
$ mv opencv_contrib-4.5.0 opencv_contrib
$ cd opencv
$ mkdir build
$ cd build
$ cmake -D CMAKE_BUILD_TYPE=RELEASE         -D CMAKE_INSTALL_PREFIX=/usr/local         -D OPENCV_EXTRA_MODULES_PATH=~/opencv_contrib/modules         -D ENABLE_NEON=ON         -D BUILD_TIFF=ON         -D WITH_FFMPEG=ON         -D WITH_GSTREAMER=ON         -D WITH_TBB=ON         -D BUILD_TBB=ON         -D BUILD_TESTS=OFF         -D WITH_EIGEN=OFF         -D WITH_V4L=ON         -D WITH_LIBV4L=ON         -D WITH_VTK=OFF         -D OPENCV_ENABLE_NONFREE=ON         -D INSTALL_C_EXAMPLES=OFF         -D INSTALL_PYTHON_EXAMPLES=OFF         -D BUILD_NEW_PYTHON_SUPPORT=ON         -D BUILD_opencv_python3=TRUE         -D OPENCV_GENERATE_PKGCONFIG=ON         -D BUILD_EXAMPLES=OFF ..
$ cd ~/opencv/build
$ make -j4
$ sudo make install
$ sudo ldconfig
$ make clean
$ sudo apt-get update
```
Turn off Raspberry Pi, take out the microSD card and edit the config.txt in system-boot section.
add start_x=1 before the enable_uart=1 line.

```bash
$ sudo apt install ffmpeg
$ ffmpeg -f video4linux2 -s 640x480 -i /dev/video0 -ss 0:0:2 -frames 1 capture_test.jpg
```

Hereisemi, 2022-05-30:
Thanks your for update!

I'm pretty new to all of this and im not 100% sure what to do now.

So you are building the OpenCR on the RPi3B+, why are you doing this exactly? As i recall, i had to download/setup the OpenCR files to the RPi3B+ and then upload the files to the OpenCR. This was the step in the "OpenCR Setup" in the e-Manual (https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup). 
Or is yours just an other possibilty to set up the OpenCR?

To be honest im scared to mess my RPi3B+ up, I'm pretty happy that it kinda works right now (except of the camera). That is why I want to be sure to not mess it up.

Should I do your OpenCV building first and then try the commands:

> "Turn off Raspberry Pi, take out the microSD card and edit the config.txt in system-boot section.
> add start_x=1 before the enable_uart=1 line."
> 
> $ sudo apt install ffmpeg
> $ ffmpeg -f video4linux2 -s 640x480 -i /dev/video0 -ss 0:0:2 -frames 1 capture_test.jpg

or can I try this ^ right now with my OpenCR and RPi3B+ setup?

ROBOTIS-Will, 2022-05-31:
@Hereisemi 
The OpenCR firmware doesn't need to be built.
The process that I mentioned above is to install necessary packages/libraries and to get an image from the camera.
I've run the last command and successfully received "capture_test.jpg" from the camera.
Since I confirmed that the RPI V2 camera works fine, I'm going to write the camera node for ROS2 Foxy.

Hereisemi, 2022-05-31:
Alright, thanks!

all your try your mentioned steps above as soon as I'm at home (tomorrow):

> $ sudo apt-get install build-essential cmake gcc g++ git unzip pkg-config
> $ sudo apt-get install libjpeg-dev libpng-dev libtiff-dev libavcodec-dev libavformat-dev libswscale-dev libgtk2.0-dev [...]

> Turn off Raspberry Pi, take out the microSD card and edit the config.txt in system-boot section.
> add start_x=1 before the enable_uart=1 line.

> $ sudo apt install ffmpeg
> $ ffmpeg -f video4linux2 -s 640x480 -i /dev/video0 -ss 0:0:2 -frames 1 capture_test.jpg

I'll keep you updated :)

ROBOTIS-Will, 2022-06-02:
@Hereisemi 
Alright, once I complete the test, I'll share the SD card image of mine so that you can try on yours.
Building and installing the source code takes quite long and sometimes the RPi seemed to be frozen, but it actually was running the command.
So I just ran the command and left it overnight.

Hereisemi, 2022-06-04:
It might be a problem with the software on the RaspberryPi. I installed the SBC Software from the tutorial (https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup) (Foxy -> [Download Raspberry Pi 3B+ ROS2 Foxy image]) and i think its a preinstalled Server Ubuntu 20.04 on arm64 and not a "Raspbian distributed by ROBOTIS". 

The Raspberry Pi Camare seems to have issues with the arm64 version, but i havent found any workarounds yet. 

Did you install it on this exactly same software like me?

spragunr, 2022-07-05:
I've been able to get the camera working on the waffle-pi under Foxy and have submitted a pull request #882 with the supporting ROS code. I didn't need to build anything, but did need to install some packages:
```
sudo apt install ros-foxy-image-tools
sudo apt install ros-foxy-usb-cam
sudo apt install ros-foxy-compressed-image-transport
```
And reboot after modifying `/boot/firmware/config.txt` as discussed above.

Hereisemi, 2022-08-01:
> I've been able to get the camera working on the waffle-pi under Foxy and have submitted a pull request #882 with the supporting ROS code. I didn't need to build anything, but did need to install [...]

Thanks for your reply and sry for my late response. 
I just started to work on my turtlebot3 again and I'm not sure how to proceed here. I looked at your pull request and tried to import your changed in the "turtlebot3_bringup" folder, but i cant find the folder "turtlebot3_bringup" on the Raspberry PI. Do you have an exact path for me?

Did you install the packages above on your Remote PC, right?

Thanks!

spragunr, 2022-08-01:
Those packages should be installed both on the remote PC and the Raspberry Pi. The `turtlebot3_bringup` folder is probably located under `~/turtlebot3_ws/src/turtlebot3` on the Raspberry Pi.  I should have been more clear when I said "I didn't need to build anything".  I meant that I didn't need to build any of the dependencies. You'll need to colcon build the turtlebot3 packages on both the pi and the remote computer (at least `turtlebot3_bringup`) to use the updated code from the pull request.

Hereisemi, 2022-08-01:
Thanks for your fast reply! I found the folder and I edited / created the file you mentioned in your pull request. I also used the 3 command to install the files on both (RBP and Remote PC).

[image]

I rebooted the RBP and im getting an error when i try to execute the "ros2 launch turtlebot3_bringup robot.launch.py" command.

[image]

[image]

Hereisemi, 2022-08-01:
Why is it looking in this directory? 
[image]

The edited files have this path "/home/ubuntu/turtlebot3_ws/src/turtlebot3/turtlebot3_bringup/camera_info"

Hereisemi, 2022-08-01:
I fixed it. I put the file in the "share" path aswell and now its running again with some additional camera info.

But i still get 1 error in the terminal

[image]

spragunr, 2022-08-15:
You may need to add your user to the `video` group on the pi:
```
sudo usermod -a -G video youruser
```

limenutt, 2022-09-10:
@Hereisemi, I think I'm having a similar issue with Turtlebot3 Waffle PI with Raspberry Pi 4 and Camera Module V2. 
Please let me know if you have solved it since then. I've had a bit of progress, but still can't get the camera to work.

I've updated the code following PR, rebuilt the turtlebot3_ws and run into an error on bringup `Cannot identify /dev/device0`.
Indeed when I checked `ls -l /dev | grep video` there is no `/dev/device0`, instead there are many other video devices from `/dev/video10` to `/dev/video16`. 
I've tried all of them by updating the `video_device` setting in  `turtlebot3_ws/src/turtlebot3/turtlebot3_bringup/param/camera.yaml` and I've got some progress with `/dev/video14`

I now have a different error on bringup, it seems that it can connect to the camera, but the camera doesn't support the capturing mode that usb_cam is trying to use. 
Here is what I'm getting 

```[usb_cam_node_exe-2] [INFO] [1662775569.913041039] [camera.usb_cam]: Starting 'camera' (/dev/video14) at 640x480 via mmap (yuyv) at 10 FPS
[usb_cam_node_exe-2] [ERROR] [1662775569.913590205] [usb_cam]: can't set stream params 25
[usb_cam_node_exe-2] [INFO] [1662775569.913663372] [usb_cam]: This Cameras Supported Formats:
[usb_cam_node_exe-2] [INFO] [1662775569.913721131] [usb_cam]:   YUYV 4:2:2[Index: 0, Type: 1, Flags: 0, PixelFormat: 56595559]
[usb_cam_node_exe-2] [INFO] [1662775569.913761039] [usb_cam]:   width: 64 x height: 16384
[usb_cam_node_exe-2] [INFO] [1662775569.913800687] [usb_cam]:   YVYU 4:2:2[Index: 1, Type: 1, Flags: 0, PixelFormat: 55595659]
[usb_cam_node_exe-2] [INFO] [1662775569.913831002] [usb_cam]:   width: 64 x height: 16384
[usb_cam_node_exe-2] [INFO] [1662775569.913868335] [usb_cam]:   VYUY 4:2:2[Index: 2, Type: 1, Flags: 0, PixelFormat: 59555956]
[usb_cam_node_exe-2] [INFO] [1662775569.913898076] [usb_cam]:   width: 64 x height: 16384
[usb_cam_node_exe-2] [INFO] [1662775569.913935909] [usb_cam]:   UYVY 4:2:2[Index: 3, Type: 1, Flags: 0, PixelFormat: 59565955]
[usb_cam_node_exe-2] [INFO] [1662775569.913965242] [usb_cam]:   width: 64 x height: 16384
[usb_cam_node_exe-2] [INFO] [1662775569.914002539] [usb_cam]:   Planar YUV 4:2:0[Index: 4, Type: 1, Flags: 0, PixelFormat: 32315559]
[usb_cam_node_exe-2] [INFO] [1662775569.914032316] [usb_cam]:   width: 64 x height: 16384
[usb_cam_node_exe-2] [INFO] [1662775569.914069705] [usb_cam]:   Planar YVU 4:2:0[Index: 5, Type: 1, Flags: 0, PixelFormat: 32315659]
[usb_cam_node_exe-2] [INFO] [1662775569.914101705] [usb_cam]:   width: 64 x height: 16384
[usb_cam_node_exe-2] [INFO] [1662775569.914140724] [usb_cam]:   24-bit RGB 8-8-8[Index: 6, Type: 1, Flags: 0, PixelFormat: 33424752]
[usb_cam_node_exe-2] [INFO] [1662775569.914171076] [usb_cam]:   width: 64 x height: 16384
[usb_cam_node_exe-2] [INFO] [1662775569.914208613] [usb_cam]:   24-bit BGR 8-8-8[Index: 7, Type: 1, Flags: 0, PixelFormat: 33524742]
[usb_cam_node_exe-2] [INFO] [1662775569.914238946] [usb_cam]:   width: 64 x height: 16384
[usb_cam_node_exe-2] [INFO] [1662775569.914277502] [usb_cam]:   32-bit BGRA 8-8-8-8[Index: 8, Type: 1, Flags: 0, PixelFormat: 34325241]
[usb_cam_node_exe-2] [INFO] [1662775569.914309427] [usb_cam]:   width: 64 x height: 16384
[usb_cam_node_exe-2] [INFO] [1662775569.914347927] [usb_cam]:   16-bit RGB 5-6-5[Index: 9, Type: 1, Flags: 0, PixelFormat: 50424752]
[usb_cam_node_exe-2] [INFO] [1662775569.914378872] [usb_cam]:   width: 64 x height: 16384
[usb_cam_node_exe-2] [INFO] [1662775569.914416705] [usb_cam]:   Y/CbCr 4:2:0[Index: 10, Type: 1, Flags: 0, PixelFormat: 3231564e]
[usb_cam_node_exe-2] [INFO] [1662775569.914447594] [usb_cam]:   width: 64 x height: 16384
[usb_cam_node_exe-2] [INFO] [1662775569.914486224] [usb_cam]:   Y/CrCb 4:2:0[Index: 11, Type: 1, Flags: 0, PixelFormat: 3132564e]
[usb_cam_node_exe-2] [INFO] [1662775569.914517539] [usb_cam]:   width: 64 x height: 16384
[usb_cam_node_exe-2] [INFO] [1662775569.914692631] [camera.usb_cam]: starting timer 100
```
I have no idea how to reconfigure the usb_cam to use one of those settings :( 
@spragunr maybe you could help?

limenutt, 2022-09-12:
I've managed to get camera working on my RPI4 8Gb + Camera module V2.1 + ROS2 Foxy setup

Here is the step-by-step guide for those who'll face the same problem:

0) make sure your camera is plugged into a camera slot, not the display slot on RPI! (took me a while to notice it)
1) burn Ubuntu Server 20.04 **arm 64bit** image onto SD card — the image provided on the Turtlebot3 Foxy quickstart guide didn't work for me
2) connect RPI4 to display and keyboard and run through the initial setup: 
- set ubuntu user password (default password is `ubuntu` and it asks you to reset password on first login)
- setup networking in /etc/netplan/50-cloud-init.yaml, then run `sudo netplan apply`. In my case, I've set up dhcp for WiFi and static IP for ethernet to be able to connect to the Turtlebot3 from my laptop.
3) make changes to /boot/firmware/config.txt by adding 2 lines at the end
```
start_x=1
gpu_mem=128
```
4) `sudo apt-get update && sudo apt-get upgrade`
5) `sudo apt-get install v4l-utils` and then `sudo modprobe bcm2835-v4l2`
6) reboot and reconnect via SSH
7) test that the camera now works:
- `ls -l /dev | grep video` should show */dev/video0*
- take a picture
```
v4l2-ctl --set-fmt-video=width=2592,height=1944,pixelformat=3
v4l2-ctl --stream-mmap=3 --stream-count=1 --stream-to=somefile.jpg
```
if the camera doesn't work at this point, you can try the following workarounds:

**Ensure your user has access to the camera device**
```
sudo usermod -aG video $USER
sudo chmod 777 /dev/vchiq
```

**Update your RPI firmware**
```
sudo curl -L --output /usr/bin/rpi-update https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update && sudo chmod +x /usr/bin/rpi-update
sudo rpi-update
```
then reboot

After verifying that the camera works fine, follow the instructions from this thread:

8) Install additional packages for camera on **both** RPI and device you'll be running rviz from (a.k.a PC)
```
sudo apt install ros-foxy-image-tools
sudo apt install ros-foxy-usb-cam
sudo apt install ros-foxy-compressed-image-transport
```
9) apply changes from [this PR](https://github.com/ROBOTIS-GIT/turtlebot3/pull/882/files) to files in `~/turtlebot3_ws/src` and then rebuild by running `colcon build` inside `~/turtlebot_ws`
10) follow the [Quickstart guide from Bringup onwards](https://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#bringup) 

**CREDIT**

I'd like to credit several sources that were very useful in solving this problem:
1) @spragunr's comments here and their [PR](https://github.com/ROBOTIS-GIT/turtlebot3/pull/882/files) 
2) @jordanmack [comment](https://github.com/raspberrypi/userland/issues/314#issuecomment-554610215) on getting /dev/video0 device to work and taking a still picture without RPI native tools
3) [blob post](https://zengliyang.wordpress.com/2021/01/04/raspberry-pi-4b-ubuntu-20-04-camera/) by  on getting camera to work on Ubuntu 20.04 armhf (32bit arm)

Hereisemi, 2022-09-14:
Hey @limenutt,
no unfortunately not. But i haven't worked on it since then.

Thanks for your guide! I did just try it on my already modified version but i always get this error message:

[image]

I guess its probably my "old" version on which i already tried several solutions without luck or it is the different Ubuntu Server 20.04 arm 64bit image which you are mentioning in your first step. I always tried it with the image provided by the quickstart guide.

I'll try your guide on a fresh installed image now.

limenutt, 2022-09-14:
@Hereisemi 
> Thanks for your guide! I did just try it on my already modified version but i always get this error message:
yeah, I've also struggled to get the /dev/video0 device to show up, I think these steps are required for this
1. adding these lines to the end of ` /boot/firmware/config.txt` — please note the folder here is **/boot/firmware**, not **/boot** (there is another config.txt file in the /boot folder but it's not used) and you need to do it by connecting via ssh to RPI, and not by modifying the file by connecting sd card to a laptop.
```
start_x=1
gpu_mem=128
```
2. `sudo modprobe bcm2835-v4l2` 
this is also crucial, without it I couldn't get /dev/video0 to appear
3.  reboot

Let me know how it goes.

Hereisemi, 2022-09-16:
@limenutt 
The camera is showing up now at least! It turned it to be a hardware issue. I changed the camera cable and i can pictures now... finally.

But the issue with the cartographer is still there. 

I did your step 8 on the RBP and my Remote PC. I also applied all the changes from the Pull Request and rebuild it. The new build took about 22 minutes.

The good point is that the camera is finally properly connected and set up, but the cartographer still had some issues.

Is your camera in the cartographer right there at the beginning or do you have to use the "add" button to add the camera?

limenutt, 2022-09-16:
@Hereisemi If you want to see the camera input in rviz, you need to add the **Image** from the `/camera/uncompressed` topic.

Hereisemi, 2022-09-19:
> @Hereisemi If you want to see the camera input in rviz, you need to add the **Image** from the `/camera/uncompressed` topic.

It's working! But it's really slowing down my the whole Raspberry Pi. If I'm lucky i will receive one picture and then the whole data stream starts the delay and lag. I guess the uncompressed image data is too much for the Raspberry PI.

[image]

How does it run on your Raspberry Pi? Do you have a smooth camera stream?

limenutt, 2022-09-19:
Great to hear it's working @Hereisemi !
How are you connecting to the RPI? There could be congested network, you could try changing camera resolution. I also have some video lags when I connect over WiFi, but it doesn't affect the cartographer's performance on the map building (I guess it needs at least 1 FPS and I think I can get around 10-13 FPS).

Hereisemi, 2022-09-19:
Update!
@limenutt yes it is actually the WiFi which struggles. Probably some issue with some ports and handling some of the packages.

I tried it with an Ethernet connection and it worked perfectly!

Do you know by chance which Port(s) the RViz uses? I want to try to fix the WiFi issue as well. I tried to check with WireShark, but couldn't find any useful data. The data throughput with the camera and movement is only at about 18 MiB/s, which is kinda low. The WiFi shouldn't struggle with this amount.

Any idea?

Anyways, thank you very much for your help on this topic! @limenutt

limenutt, 2022-09-20:
@Hereisemi , ROS2 is using different ports depending on your ROS Domain ID (you set it up with the enviroment variable, by default for turtlebot3 it's `30`).
If you put your domain id [here](https://docs.ros.org/en/galactic/Concepts/About-Domain-ID.html#domain-id-to-udp-port-calculator) in this form you'll see what ports are used in your case. 
With WiFi it could be not an issue of volume but _speed_. So maybe check your ping, your router may also contribute to it.
Think about it this way:  say, you're transferring water from one place to another and have a very wide pipe but a very long one. Even if you are moving a small amount of water over the long pipe it doesn't matter how wide the pipe is, what matters is how long is it and how fast it's moving.. I hope it makes sense :)

Hereisemi, 2022-09-21:
@limenutt yes, that makes sense. You are right with the speed. I just tested the "image" while pinging in both directions. The first couple pings are without the "image" and then I added the "image" and the ping went up.

[image]

Do you think I can solve this issue by opening up the ports in the router?

limenutt, 2022-09-21:
The ports should be already open otherwise nothing would work at all :)
You can try updating compression settings to make images smaller, it could
help.

Hereisemi, 2022-09-21:
Yea true, maybe there is a restriction or some other kind of issue. I'll try it on a different router soon and see how it runs.

Can you tell me where I can modify the camera settings? 

I follow the path while the bringup command starts and i found the "turtlebot3_rpicamera.yaml" file. I tried to modify the file on the turtlebot, but it seems that it always gets overwritten.

I tried to reduce the resolution to 160x120 but it still started with 640x480. I can't find a setting for the framerate either.

Is there a easier way to modify the resolution and FPS?

[image]

Hereisemi, 2022-09-21:
With the reduction to 160x120 and 5 FPS the transfer dropped from 18 MiB/s to 400 KiB/s as expected and now its working on WiFi as well. I'll test it on a other router soon and I'll play around with the camera setting to see where to bottleneck is.

Thank you again for all of your help @limenutt !

limenutt, 2022-09-21:
Glad to hear it works now!

sunghowoo, 2024-11-13:
This issue appears to be resolved, so it will be closed.

closed 2024-11-13 by sunghowoo