ROBOTIS-GIT/turtlebot3 #865: issue with odometry not following commanded velocity angular velocity unequal zero with only x-velocity commanded

florianfhws, 2022-05-24:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [x ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ x] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - using the teleop script to drive the robot in x direction
    -> odometry velocities for angular velocity are unequal to zero
    -> odometry values for velocities are not zero when standing still

8. Copy and Paste the error messages on terminal.

see above

9. Please describe the issue in detail.

The odometry values should be zero for velocities that are not commanded

ROBOTIS-Will, 2022-05-25:
Hi @florianfhws 
Have you stopped the robot by pressing the `space` or `s` key?
https://github.com/ROBOTIS-GIT/turtlebot3/blob/melodic-devel/turtlebot3_teleop/nodes/turtlebot3_teleop_key#L162

florianfhws, 2022-05-25:
@ROBOTIS-Will 
This issue occures right from the start. The angular velocity above is shown even so the robot was just started up and standing still, the value however changes permanently.
When we start moving with just /cmd_vel x being 0.1 and all other values 0, the angular velocity still shows this behavior.

closed 2024-11-15 by sunghowoo