ROBOTIS-GIT/turtlebot3 #870: Navigation Issue in Noetic

pradrn, 2022-06-01:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/map.yaml

8. Copy and Paste the error messages on terminal.

[image]
[image]

9. Please describe the issue in detail.

[image]

So when i was running the Navigation Tutorial, I was not getting the LDS sensor overlay on my map. And i was continuously getting warnings regarding Costmap2ROS as you can see in the images present. Please help in solving this issue for me.

ROBOTIS-Will, 2022-06-02:
Hi @pradrn 
There is an identical issue #78.
Does it help to solve your problem?

pradrn, 2022-06-02:
Hi,

So i did have a look at the identical issue #78 as mentioned. I am yet to try fiddling around with chrony config file and DS3231.

This is the frequency of the tf topics which i checked
[image]

I tried to sync the time between the remote laptop and the RasPi. I ran the command in both remote PC and the RasPi
[image]
[image]

I then used the command `sudo date -s "dd mm yyy hh:mm:ss` to update the time manually. Then i ran the navigation command again, yet the same issue continued. The LDS overlap was missing and the Costmap2ROS warning were present.

[image]

BENBEY-I, 2022-06-03:
i had the same problem, here's the steps that i did to solve it : 
- deactivated the automatic date and time in the PC setting 
- changed the time ( -2h to have UTC time)
try this 2 steps if it doesn't work try to change the time in the raspberry using the command sudo date -s "dd mm yyy hh:mm:ss"

i solved it just by setting the time

pradrn, 2022-06-06:
Hi @BENBEY-I,

Thanks for the response. I tried the steps you mentioned and disabled the automatic date and time in remote PC, changed the time in both RasPi and PC so that  they match. The warnings and LDS overlay did not occur.

In addition to this, next I played  around with the  RTC to make sure that it is synced with the local time and TZ. After this I reenabled the NTP in  both remote and RasPi.

Unfortunately, I was still not  able to get the LDS overlay and the warnings  for Costmap2DROS was present.

sunghowoo, 2024-12-06:
@pradrn Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-12-06 by sunghowoo