ROBOTIS-GIT/turtlebot3 #876: Turtlebot3 Won't Connect to School WiFi

dean24-up, 2022-06-15:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ **X** ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ **X** ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ **X** ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ **X** ] etc (ROS Noetic image distributed by ROBOTIS)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ **X** ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

We're following the ROBOTIS Turtlebot3 e-Manual Noetic instructions for SBC Setup, and we ran into an issue with step 3.2.6 with editing the SD card. The network we want to connect the Turtlebot to is a school network that requires both a username and a password, but the instrcutions only show how to edit 50-cloud-init.yaml by adding a password. We  added an identity line for the username, but it didn't seem to help.

All of our alternatives for WiFi don't require usernames or passwords at all, like school Guest or IoT networks. We couldn't connect the Turtlebot3 through any of these networks using the SD card setup. We've made a little progress with Turtlebot3 hardware installation and setup by connecting the RPi to ethernet, but we need WiFi to be able to connect to the RemotePC.

ROBOTIS-Will, 2022-06-16:
Hi @dean24-up 

For a specific network environment, please contact your network manager.
You may refer to the link below about specifying the username and password.
https://askubuntu.com/questions/1352808/how-to-connect-raspberry-pi-to-wifi-that-requires-username-and-password

Thank you.

dean24-up, 2022-06-16:
Hi @ROBOTIS-Will 

Thank you! We followed the instructions in the link and got the username/password network to work. We'll contact our network manager about the other networks, but the TurtleBot is now connected to WiFi.

ROBOTIS-Will, 2022-06-17:
@dean24-up 

Awesome! Thanks for the udpates :)
I'll close this ticket if you don't have additional issues related to this.

closed 2022-06-17 by ROBOTIS-Will