ROBOTIS-GIT/turtlebot3 #878: Having trouble to bring up

Keita1Mori, 2022-06-20:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [x] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    

8. Copy and Paste the error messages on terminal.

    [image]

9. Please describe the issue in detail.

    -

ashekim, 2022-06-20:
Hi @Keita1Mori.
Thank you for your questions.

Please reset OpenCR by referring to the attached [link](https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup).
If the same problem still occurs, please replace the USB cable.

Thank you.

Keita1Mori, 2022-06-20:
Thanks.
I will try it.

Keita1Mori, 2022-06-21:
Couldn't update on Ubuntu mate
[image]

ashekim, 2022-06-22:
Hi @Keita1Mori 
Please check if the Rpi network is connected.
I'm attaching a [link](https://askubuntu.com/questions/91543/apt-get-update-fails-to-fetch-files-temporary-failure-resolving-error) with a problem similar to yours.

Additionally,
We are providing the SBC image for the convenience of users.
We recommend that you use the [images](https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup) we provide.

Thank you.

Keita1Mori, 2022-06-24:
I'm still having trouble for opencr setting I tried the switching but still it says NG. What is wrong with this?
[image]

ROBOTIS-Will, 2022-06-24:
@Keita1Mori 

It looks like you didn't configured the serial port access for OpenCR.
Please fix the permission of the OpenCR connected port (in Ubuntu, it is /dev/ttyACM0)
```bash
$ sudo usermod -aG dialout turtlebot3
```
then reboot to apply the permission.

Keita1Mori, 2022-06-24:
Thanks I have solved that problem but I still have another problem with this can you help me to solve it?
[image]

ashekim, 2022-06-27:
@Keita1Mori 

Hi.
Please check if the launch file of the package to be executed exists.
And build again after installing the package.

Thank you.

sunghowoo, 2024-11-15:
@Keita1Mori This issue will be closed due to prolonged inactivity. If the issue persists, feel free to reopen it.

closed 2024-11-15 by sunghowoo