ROBOTIS-GIT/turtlebot3 #884: AMCL failing when starting Navigation Simulation

johnpenright, 2022-07-04:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ X] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [X] ROS2 Humble Hawksbill
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ X] Gazebo Simulation
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ X] Gazebo Simulation
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] Windows 10 + WSL2 + Ubuntu 22.04 (Jammy)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    Running Gazebo simulation tutorials (https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/). Following _Foxy_ instructions, adapted for _Humble_. The Gazebo process loads fine,

`ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py`

 but starting the Navigation node gives constant errors. 

`ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=True map:=$HOME/map.yaml`

8. Copy and Paste the error messages on terminal.

`
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_isolated-1]: process started with pid [23605]
[INFO] [rviz2-2]: process started with pid [23607]
[rviz2-2] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-XXXX'
[component_container_isolated-1] [INFO] [1656947648.761198200] [nav2_container]: Load Library: /opt/ros/humble/lib/libmap_server_core.so
[component_container_isolated-1] [INFO] [1656947648.772121000] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapSaver>
[component_container_isolated-1] [INFO] [1656947648.772207400] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapServer>
[... 64 lines cut ...]
[component_container_isolated-1] [INFO] [1656947648.979922200] [amcl]: initTransforms
[component_container_isolated-1] [INFO] [1656947649.000404200] [planner_server]:
[component_container_isolated-1]        planner_server lifecycle node launched.
[component_container_isolated-1]        Waiting on external lifecycle transitions to activate
[component_container_isolated-1]        See https://design.ros2.org/articles/node_lifecycle.html for more information.
[component_container_isolated-1] [INFO] [1656947649.002212500] [planner_server]: Creating
[component_container_isolated-1] [INFO] [1656947649.002235100] [amcl]: initPubSub
[component_container_isolated-1] [INFO] [1656947649.016361600] [amcl]: Subscribed to map topic.
[component_container_isolated-1] [ERROR] [1656947649.025091400] []: Caught exception in callback for transition 10
[component_container_isolated-1] [ERROR] [1656947649.025157400] []: Original error: According to the loaded plugin descriptions the class differential with base class type nav2_amcl::MotionModel does not exist. Declared types are  nav2_amcl::DifferentialMotionModel nav2_amcl::OmniMotionModel
[component_container_isolated-1] [WARN] [1656947649.025191400] []: Error occurred while doing error handling.
[component_container_isolated-1] [FATAL] [1656947649.025208800] [amcl]: Lifecycle node amcl does not have error state implemented
[component_container_isolated-1] [ERROR] [1656947649.025527500] [lifecycle_manager_localization]: Failed to change state for node: amcl
[component_container_isolated-1] [ERROR] [1656947649.025559700] [lifecycle_manager_localization]: Failed to bring up all requested nodes. Aborting bringup.`

9. Please describe the issue in detail.

The above failure in configuring AMCL seems to be the key issue. Here are a few other observations: 

- The rViz window appears but does not show the map. 
- Then I get a constant flow of error messages: 

`[component_container_isolated-1] [INFO] [1656947652.365148100] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist`

- Using the view_frames tool, the `map` node is missing

johnpenright, 2022-07-05:
So I've made some progress. In the robot param file (`waffle_pi.yaml`), if I change line 29 from

`robot_model_type: "differential"`

to

`robot_model_type: "nav2_amcl::DifferentialMotionModel"`

then rViz will launch and initialize properly. 

However I don't seem to be getting a local costmap, even though rViz is confugured to display one. I'm not seeing any clear errors in the logs.

quyetbt2000, 2022-08-24:
Hi, Thanks for your solution, I'm in the same issue, any explanation for that change?
After following your instruction, any solution to get a local costmap back ?

johnpenright, 2022-08-24:
> Hi, Thanks for your solution, I'm in the same issue, any explanation for that change? After following your instruction, any solution to get a local costmap back ?

I don't think I needed to do anything specific; I just switched over to the _cyclonedds_.

quyetbt2000, 2022-08-25:
I got
 "Message Filter dropping message: frame 'base_scan'/'odom' at time 24.074 for reason 'discarding message because the queue is full'". Is it any way to fix it in src file, and not switch to cyclonedds ( I have no idea about this)

ncdejito, 2022-09-07:
switch to cyclonedds with
```
sudo apt install ros-<ros-distro>-rmw-cyclonedds-cpp
vim ~/.bashrc
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp # insert this line
source ~/.bashrc
```
when you do `ros2 launch`, check on separate terminal using `ps -eo cmd`, should show cyclonedds in `--rmw-implementation`:
```
21318 22.0  0.6 535896 55044 pts/8    Sl   16:14   0:00 /usr/bin/python3 /opt/ros/humble/bin/_ros2_daemon --rmw-implementation rmw_cyclonedds_cpp --ros-domain-id 22
```

[source](https://docs.ros.org/en/foxy/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html#install-packages)

sea-bass, 2022-10-29:
> 

> So I've made some progress. In the robot param file (`waffle_pi.yaml`), if I change line 29 from
>
> `robot_model_type: "differential"`
>
> to
>
> `robot_model_type: "nav2_amcl::DifferentialMotionModel"` [...]

Just a heads-up : Even with CycloneDDS, this failed until I fixed the AMCL configuration as above ^^^

According to [the official Nav2 documentation](https://navigation.ros.org/configuration/packages/configuring-amcl.html), we should be using `nav2_amcl::DifferentialMotionModel` anyway, so I would request this be updated.