ROBOTIS-GIT/turtlebot3 #886: Can't Run Turtlebot3 Teleop with WSL

dean24-up, 2022-07-07:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ X ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ X ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ X ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ X ] etc (ROS Noetic Image Distributed by ROBOTIS)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ X ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

I have been controlling the Turtlebot3 with a Linux machine up until now, but I've also been developing code using ROS on a Windows machine with Ubuntu 20.0.4 installed through WSL. I've been able to run Gazebo, Turtlesim, and run ROS programs well through WSL, but I haven't figured out how to run Turtlebot3 teleop. 

I can ssh into the Turtlebot3 and run bringup using Ubuntu through WSL (I can also ping websites while ssh'ed into the Turtlebot3), but when I launch teleoperation in a separate terminal with `roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch`, I get 

```
RLException: Unable to contact my own server at [http://<ip_address>].
This usually means that the network is not configured properly.

A common cause is that the machine cannot connect to itself.  Please check
for errors by running:

        ping <ip_address>

For more tips, please see

        http://wiki.ros.org/ROS/NetworkSetup

The traceback for the exception was written to the log file
```

When formatting each of the bashrc files, I tried setting the Remote PC address to the IP address that is listed by `eth0` when I run `ip a`. I also tried the IP address of my actual device on my WiFi network (outside WSL). Using my WiFi IP results in the above error, and the error from using `eth0 `is very similar. Pinging the IP address is unsuccessful in both cases.

When I run `iwconfig` in the Remote PC, no wireless extensions are listed. I suspect that the issue is that WSL doesn't allow Ubuntu access to my device's networks, but I'm confused as to why ssh still seems to work even though I can't launch teleop. Is it possible to control Turtlebot using WSL, or is it necessary to use a Linux machine?

ROBOTIS-Ashe, 2022-07-26:
The VM could cause many unexpected issues including the network configuration.
Unfortunately, VM is not officially supported by Turtlebot3.
For smooth operation, it is recommended to run Ubuntu instead of a VM.
Thank you.

dean24-up, 2022-07-26:
Hi @ROBOTIS-Ashe ,
Thanks for the clarification :)

closed 2022-07-26 by dean24-up