ROBOTIS-GIT/turtlebot3 #889: Humble Navigation simulation dropping message error

WonsokYoo, 2022-07-19:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [v ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [v ] etc (ROS 2 Humble Hawksbill)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [v ] etc (only doing simulation)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [v ] etc (only doing simulation)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [v ] etc (Ubuntu 22.04 docker container)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
    - ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=True map:=$HOME/map.yaml
    - Of course I did 'export TURTLEBOT3_MODEL=burger' on each terminal.

8. Copy and Paste the error messages on terminal.

    - [component_container_isolated-1] [INFO] [1658203409.501428806] [controller_server]: Activating
[component_container_isolated-1] [INFO] [1658203409.501464578] [local_costmap.local_costmap]: Activating
[component_container_isolated-1] [INFO] [1658203409.501513153] [local_costmap.local_costmap]: Checking transform
[component_container_isolated-1] [INFO] [1658203409.501609676] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-1] [INFO] [1658203410.001659488] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-1] [INFO] [1658203410.501710559] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-1] [INFO] [1658203411.001673960] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[rviz2-2] [INFO] [1658203411.046314441] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 5105.405 for reason 'discarding message because the queue is full'
[rviz2-2] [INFO] [1658203411.238243445] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 5105.605 for reason 'discarding message because the queue is full'
[rviz2-2] [INFO] [1658203411.431021439] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 5105.805 for reason 'discarding message because the queue is full'
[component_container_isolated-1] [INFO] [1658203411.501656674] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[rviz2-2] [INFO] [1658203411.654006711] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 5106.005 for reason 'discarding message because the queue is full'
[rviz2-2] [INFO] [1658203411.845889254] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 5106.205 for reason 'discarding message because the queue is full'
[component_container_isolated-1] [INFO] [1658203412.001686864] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[rviz2-2] [INFO] [1658203412.037697341] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 5106.405 for reason 'discarding message because the queue is full'

9. Please describe the issue in detail.

    - I already built the map using 'turtlebot3_cartographer' packages in simulation mode. Gazebo works fine but when I launch as follows:
    -   ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=True map:=$HOME/map.yaml
    - It fails with dropping  messages.

WonsokYoo, 2022-07-27:
I'm not deeply understanding but the above problem is solved referencing #884
As @johnpenright mentioned, I modified robot param file as follows:

robot_model_type: "differential"

to

robot_model_type: "nav2_amcl::DifferentialMotionModel"

I'm not sure why I have to do this because I used 'humble-devel' branch.

Is this ROBOTIS's official (future) fix? or just a trick?

HuzaifaOsal, 2023-03-31:
have you found the solution ?

GyuH13, 2024-11-05:
@WonsokYoo  Thank you for answer instead ROBOTIS.
This issue was raised so long ago that it is now challenging to address.
We have updated our source and manual for fix any error, please follow manual and try again.

closed 2024-12-06 by sunghowoo