ROBOTIS-GIT/turtlebot3 #890: No data received from scan map topic

duqiyaner1234, 2022-07-19:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - After running ros2 launch turtlebot3_cartographer cartographer.launch.py rvis does not give any map data. I can see the turtle bot is moving while running Teleoperation, but map does not show. I checked scan and map topic, there is no data received. I don't see any error log, 

8. Copy and Paste the error messages on terminal.

    - I don't see any error log. Only WARN logs below
    - [occupancy_grid_node-2] [WARN] [1658271345.127284544] [occupancy_grid_node]: submap_slices and last_frame_id is empty

9. Please describe the issue in detail.

    - I tried to follow the instruction to run SLAM (https://emanual.robotis.com/docs/en/platform/turtlebot3/slam/) However, spend several days but ends of failure to render the map. No data received. I tried to update the opencr firmware to latest (https://github.com/ROBOTIS-GIT/OpenCR-Binaries/tree/master/turtlebot3/ROS2/latest) but nothing has been changed. Please advise how I can debug LDS-02 further and check if is the hardware issue. Thanks!
[image]
[image]

duqiyaner1234, 2022-07-19:
`
[INFO] [launch]: Default logging verbosity is set to INFO
/opt/ros/foxy/share/turtlebot3_cartographer/launch/cartographer.launch.py:56: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/opt/ros/foxy/share/turtlebot3_cartographer/launch/cartographer.launch.py:56: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
/opt/ros/foxy/share/turtlebot3_cartographer/launch/cartographer.launch.py:81: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/opt/ros/foxy/share/turtlebot3_cartographer/launch/cartographer.launch.py:81: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
/opt/ros/foxy/share/turtlebot3_cartographer/launch/occupancy_grid.launch.py:44: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/opt/ros/foxy/share/turtlebot3_cartographer/launch/occupancy_grid.launch.py:44: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
[INFO] [cartographer_node-1]: process started with pid [202856]
[INFO] [occupancy_grid_node-2]: process started with pid [202858]
[INFO] [rviz2-3]: process started with pid [202860]
[cartographer_node-1] [INFO] [1658274397.696643078] [cartographer_ros]: I0719 16:46:37.000000 202856 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/turtlebot3_cartographer/config/turtlebot3_lds_2d.lua' for 'turtlebot3_lds_2d.lua'.
[cartographer_node-1] [INFO] [1658274397.698223468] [cartographer_ros]: I0719 16:46:37.000000 202856 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[cartographer_node-1] [INFO] [1658274397.700785421] [cartographer_ros]: I0719 16:46:37.000000 202856 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[cartographer_node-1] [INFO] [1658274397.701803231] [cartographer_ros]: I0719 16:46:37.000000 202856 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[cartographer_node-1] [INFO] [1658274397.702620735] [cartographer_ros]: I0719 16:46:37.000000 202856 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[cartographer_node-1] [INFO] [1658274397.703627301] [cartographer_ros]: I0719 16:46:37.000000 202856 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[cartographer_node-1] [INFO] [1658274397.704367210] [cartographer_ros]: I0719 16:46:37.000000 202856 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[cartographer_node-1] [INFO] [1658274397.705366093] [cartographer_ros]: I0719 16:46:37.000000 202856 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[cartographer_node-1] [INFO] [1658274397.706382926] [cartographer_ros]: I0719 16:46:37.000000 202856 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[cartographer_node-1] [INFO] [1658274397.707346958] [cartographer_ros]: I0719 16:46:37.000000 202856 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[cartographer_node-1] [INFO] [1658274397.708181503] [cartographer_ros]: I0719 16:46:37.000000 202856 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[cartographer_node-1] [INFO] [1658274397.794283144] [cartographer_ros]: I0719 16:46:37.000000 202856 submap_2d.cc:187] Added submap 1
[cartographer_node-1] [INFO] [1658274397.795406836] [cartographer_ros]: I0719 16:46:37.000000 202856 map_builder_bridge.cc:132] Added trajectory with ID '0'.
[rviz2-3] [INFO] [1658274398.739119308] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1658274398.739504348] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[occupancy_grid_node-2] [WARN] [1658274398.773239053] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[rviz2-3] [INFO] [1658274398.791080809] [rviz2]: Stereo is NOT SUPPORTED
[occupancy_grid_node-2] [WARN] [1658274399.773260985] [occupancy_grid_node]: submap_slices and last_frame_id is empty
`

duqiyaner1234, 2022-07-20:
Now I fixed one problem. 
I found the LDS_MODEL value in my SBC's bashrc file is LDS-01. After updating it to LDS-02 I can now see the /map topic is publishing data. But /scan topic is still empty.
From there, I have another issue. When I try to run  
ros2 launch turtlebot3_cartographer cartographer.launch.py, map is still no show, and if I change frame from map to any other value (like base_scan), riviz crashes. I see ERROR like below keep populating:
[cartographer_node-1] Error:   TF SELF_TRANSFORM: Ignoring transform from authority "Authority undetectable" with frame_id and child_frame_id  "odom" because they are the same
I attached log file. 
[Uploading tmplog.txt…]()

ROBOTIS-Will, 2022-07-21:
Hi @duqiyaner1234 

Could you check the QoS setting of the scan in the RViz?
Try switching from Best effort to Reliable or the other around.

closed 2024-11-15 by sunghowoo