ROBOTIS-GIT/turtlebot3 #891: New LDS-02 problem with measures count

dimadobriy, 2022-07-20:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ X] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ X] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ X] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ X] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

```[ERROR] [1658325094.660470]: bad callback: <bound method ROS_preset.laser_scan_callback of <main.ROS_preset object at 0x7fad44163580>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "./real_exp.py", line 594, in laser_scan_callback
    new_blocks, LL, CC, x, y = self.obstacles_parser.get_obstacles(np.array(dt.ranges), fillna='else', state=self.new_state)
  File "./real_exp.py", line 280, in get_obstacles
    degrees = degrees[~nans]
IndexError: boolean index did not match indexed array along dimension 0; dimension is 360 but corresponding boolean dimension is 244
```

10. Please describe the issue in detail.

Hi there!

Thank you so much for the support you give to the users. It's time for me to reach out to you.

I have worked with TB3 before with the LDS01 version of the laser rangefinder and everything was fine.
We bought new robots and had some problems with the new version of the laser rangefinder. The thing is that every time (measure) a different number of measurements come in, according to the documentation the resolution = 1 deg. 

So, we have three TB3s with the old lidar and five pieces TB3 with the new ones. 
On the old one it works, on the new one it doesn't.
The drivers have been updated, the bashrc file has been updated according to the tutorial. 
Nevertheless, the problem remains. 
Is it normal now that the number of measurements is different each time or not? 
In total we have 3 TB3 on the old Lidar and 7 TB3 on the new Lidar. The tests were done on all of them.
But, with old lidar it works fine, with new one not. 

Hope for your help.

Kind regards,
Dmitrii
[bringup_msg.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/9150472/bringup_msg.txt)

dimadobriy, 2022-07-25:
@ROBOTIS-Will

Hi there,

Do you have any ideas?

ROBOTIS-Will, 2022-07-25:
Hi @dimadobriy 

I'm sorry about the delayed response.
The LDS-02 sensor has a fixed sampling frequency(2300Hz), the faster RPM will decrease the resolution (higher than 1&deg;) which will reduce the size of the array data in each rotation.
Unfortunately, the motor of the LDS-02 does not have the PWM controller which makes it difficult to control the RPM.

Thank you.

dimadobriy, 2022-07-25:
Hi @ROBOTIS-Will 

Thanks for your reply. 

I didn't modify RPM. So it seems it should by default. 
However, each rotation I receive different size of the array.  
See attach. Prev execution could be another, say, 240-241.
[image]

ROBOTIS-Will, 2022-07-26:
Hi @dimadobriy 

Since the USB2LDS does not have a feedback controller to control the RPM of the motor of the LDS-02, the sensor will rotate with whatever the duty cycle it is provided via PWM#1 of the USB2LDS.
While the motor is rotating, the sensor will collect data with 2300Hz therefore, depending the RPM of the motor, the size of the data array could vary.

The USB2LDS is configured to generate 70% duty cycle to the motor at PWM#1 pin.
In order to reduce the duty cycle to lower the RPM (which will improve the resolution and increase the size of the array), you need to adjust resistors(R16, R17).
For example, replacing the 3.9Kohm R16 to 1.5Kohm will reduce the duty cycle from 70% to 60%.

The equation of the output duty cycle is : `1 - { R17 / (R16 + 2*R17) }`

However, please be aware that this modification will void your product warranty so please perform this at your own risk.

Thank you.

dimadobriy, 2022-07-26:
Hi @ROBOTIS-Will 

Thank you for your reply. I really appreciate it.

Do I understand correctly that in this case the resolution of 1 degree is not respected? 
I made some adjustments in my code. By correcting the resolution I solved the problem. 

Thank you for your patience.

closed 2022-07-26 by dimadobriy

ROBOTIS-Will, 2022-07-26:
@dimadobriy 

Yes, unless you have a feedback controller to control the PWM signal, a exact 1&deg; resolution may be hard to achieve.
In case the RPM gets lower than 383, you will get more than 360 data per each rotation.

Thank you!

2rgosum, 2022-11-07:
@dimadobriy 

How do you solve this problem??

I have same problem.