ROBOTIS-GIT/turtlebot3 #893: Turtlebot3_burger unable to communicate with ROS Master run on remotePC

kidambikb, 2022-07-21:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ -] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ -] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [- ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ -] Ubuntu 20.04.4 LTS  (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ -] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [20.04]
    - Firmware version: [20.04]

7. Specify the commands or instructions to reproduce the issue.
     
    - 

8. Copy and Paste the error messages on terminal.

    - RLException: ERROR: unable to contact ROS master at [http://192.168.1.8:11311]
The traceback for the exception was written to the log file

9. Please describe the issue in detail.

    I have followed the exact procedure on [emanualrobotics.com ]. (https://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#bringup). When Bringingup, I launch roscore on my virtual machine and I was able to ssh into the turtlebot. Then I do the command of  roslaunch turtlebot3_bringup turtlebot3_robot.launch it doesn't do anything. After a quite a bit of time, it says RLException: ERROR: unable to contact ROS master at [http://192.168.1.8:11311]. 
I have double checked my bashrc files of remotePC and turtlebot according to the link here (https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#pc-setup).

ROBOTIS-Will, 2022-07-22:
@kidambikb 

The virtual machine could cause many unexpected issues including the network configuration.
If you are not used to ROS environment, it is strongly recommended to begin with Ubuntu first.
Please make sure that the VM has correctly configured to access the IP address and the port range that are used for ROS.
Thanks!

Azzie1101, 2022-11-06:
I have encountered the exact same problem in ubuntu. Any help/update on this issue?
Remote pc is correctly configured following the e-manual and ROS setup environment steps. 
Thank you in advance!

sunghowoo, 2024-11-15:
This issue will be closed due to prolonged inactivity. For optimal performance, please run it on a real hardware-based kernel.
If you have any further questions or need assistance, feel free to reach out.

closed 2024-11-15 by sunghowoo