ROBOTIS-GIT/turtlebot3 #897: NEED TO INCREASE RECOMMENDED SPEED OF SIMULATED TURTLEBOT3

PhilipAmadasun, 2022-07-28:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [Yes ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [Yes ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

   I need to increase the velocity and acceleration of simulated turtlebot3 to about 3m/s and 0.5 m/s respectively, is this possible? If so, what files do I need to edit? How could any edits potential break something else?

ROBOTIS-Ashe, 2022-08-09:
@PhilipAmadasun 

sorry for the late reply.
you can change the BURGER_MAX_LIN_VEL value and BURGER_MAX_ANG_VEL in the attached [link](https://github.com/ROBOTIS-GIT/turtlebot3/blob/9867e2f1f5c9ed54d2b41998a77a9c8eb42ebafd/turtlebot3_teleop/nodes/turtlebot3_teleop_key#L38).
Also, You can also modify the profile values from the [link](https://github.com/ROBOTIS-GIT/turtlebot3/blob/9867e2f1f5c9ed54d2b41998a77a9c8eb42ebafd/turtlebot3_teleop/nodes/turtlebot3_teleop_key#L184).
Thank you.

PhilipAmadasun, 2022-08-28:
@ROBOTIS-Ashe I'm not talking about just when I use teleop. I want this the turtlebot3 max speed increased even when I'm using 2D_Nav_Goal. I tried changing the linear speed in the turtlebot3_drive.h script but nothing seemed to happen

closed 2024-11-15 by sunghowoo