ROBOTIS-GIT/turtlebot3 #899: SLAM generates multiple maps.

kanpapa, 2022-08-07:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [X] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [X] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [X] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [X] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [V220624R1]

7. Specify the commands or instructions to reproduce the issue.

    - In the process of SLAM with keyboard operation.

8. Copy and Paste the error messages on terminal.

    - Error message not found.
    - See attachment for Rviz2 screen.
    [image]

9. Please describe the issue in detail.

    - When SLAM is performed, the map may suddenly move to another location. As a result, multiple maps are generated.
    - This does not occur with ROS1 Melodic.

kanpapa, 2022-08-07:
Another pattern.
[image]

kanpapa, 2022-08-07:
Continuing, three maps were created.
[image]

ROBOTIS-Ashe, 2022-08-09:
@kanpapa 

Hi. Sorry for the late reply.
The reported issue seems to have a similar cause to #880.
I'll take a look at the OpenCR firmware.
thank you

kanpapa, 2022-08-09:
Thank you for your reply.
I will also include the OpenCR update log since you mentioned a possible firmware issue.
Thank you in advance.

```
ubuntu@ubuntu:~/opencr_update$  ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr
aarch64
arm
OpenCR Update Start..
opencr_ld_shell ver 1.0.0
opencr_ld_main
[  ] file name          : burger.opencr
[  ] file size          : 132 KB
[  ] fw_name            : burger
[  ] fw_ver             : V220624R1
[OK] Open port          : /dev/ttyACM0
[  ]
[  ] Board Name         : OpenCR R1.0
[  ] Board Ver          : 0x17020800
[  ] Board Rev          : 0x00000000
[OK] flash_erase        : 1.07s
[OK] flash_write        : 1.33s
[OK] CRC Check          : D26544 D26544 , 0.004000 sec
[OK] Download
[OK] jump_to_fw
ubuntu@ubuntu:~/opencr_update$
```

kanpapa, 2022-08-10:
I checked odm and the value of position y has suddenly increased.

before
[image]

after
[image]

kanpapa, 2022-08-11:
This seems to be the same issue #821

kanpapa, 2022-11-05:
I updated the OpenCR firmware to 0.0.3 and now it works fine.
I will try to run turtlebot3 with foxy in this state for a while.

```
ubuntu@ubuntu:~$ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/0.0.3/opencr_update.tar.bz2
ubuntu@ubuntu:~$ tar -xjf ./opencr_update.tar.bz2
ubuntu@ubuntu:~$ cd opencr_update/
ubuntu@ubuntu:~/opencr_update$ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr
aarch64
arm
OpenCR Update Start..
opencr_ld_shell ver 1.0.0
opencr_ld_main 
[  ] file name   	: burger.opencr 
[  ] file size   	: 127 KB
[  ] fw_name     	: burger 
[  ] fw_ver      	: V190829R3 
[OK] Open port   	: /dev/ttyACM0
[  ]
[  ] Board Name  	: OpenCR R1.0
[  ] Board Ver   	: 0x17020800
[  ] Board Rev   	: 0x00000000
[OK] flash_erase 	: 1.06s
[OK] flash_write 	: 1.29s 
[OK] CRC Check   	: CBBCC4 CBBCC4 , 0.004000 sec
[OK] Download 
[OK] jump_to_fw 
ubuntu@ubuntu:~/opencr_update$ 
```

rageorgi, 2022-11-28:
> I downgraded the OpenCR firmware from the latest to 0.0.3 and it now works properly. I will try to run turtlebot3 with foxy in this state for a while.
> 
> ```
> ubuntu@ubuntu:~$ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/0.0.3/opencr_update.tar.bz2
> ubuntu@ubuntu:~$ tar -xjf ./opencr_update.tar.bz2
> ubuntu@ubuntu:~$ cd opencr_update/
> ubuntu@ubuntu:~/opencr_update$ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr
> aarch64
> arm
> OpenCR Update Start..
> opencr_ld_shell ver 1.0.0
> opencr_ld_main 
> [  ] file name   	: burger.opencr 
> [  ] file size   	: 127 KB
> [  ] fw_name     	: burger 
> [  ] fw_ver      	: V190829R3 
> [OK] Open port   	: /dev/ttyACM0
> [  ]
> [  ] Board Name  	: OpenCR R1.0
> [  ] Board Ver   	: 0x17020800
> [  ] Board Rev   	: 0x00000000
> [OK] flash_erase 	: 1.06s
> [OK] flash_write 	: 1.29s 
> [OK] CRC Check   	: CBBCC4 CBBCC4 , 0.004000 sec
> [OK] Download 
> [OK] jump_to_fw 
> ubuntu@ubuntu:~/opencr_update$ 
> ```

Hi, is SLAM and navigation with NAV2 working properly for you now in Foxy? Also, did you run these commands directly in Foxy?

ROBOTIS-Will, 2023-01-27:
An updated OpenCR ROS2 firmware V220127R1 (0.2.1) is released.
Please [download and upload the new firmware](https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup) and let me know if this doesn't fix the problem.
Thank you very much for your patience!

kanpapa, 2023-06-03:
The problem has been resolved in the firmware V220127R1 (0.2.1).
Thank you very much.

closed 2023-06-03 by kanpapa