ROBOTIS-GIT/turtlebot3 #910: RpLidar A1M8 stops spinning when calling roslaunch turtlebot3_bringup turtlebot3_robot.launch

DarekLin, 2022-10-02:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - ros2 launch turtlebot3_bringup robot.launch.py

8. Copy and Paste the error messages on terminal.

  ```
 ubuntu@ubuntu:~$ ros2 launch turtlebot3_bringup robot.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2022-10-02-18-28-35-412562-ubuntu-2047
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
[INFO] [robot_state_publisher-1]: process started with pid [2049]
[INFO] [ld08_driver-2]: process started with pid [2051]
[INFO] [turtlebot3_ros-3]: process started with pid [2053]
[turtlebot3_ros-3] [INFO] [1664735318.531024794] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [1664735318.536900308] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-3] [INFO] [1664735318.545365547] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [INFO] [1664735318.555682422] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link base_link had 5 children
[robot_state_publisher-1] Link caster_back_link had 0 children
[robot_state_publisher-1] Link imu_link had 0 children
[robot_state_publisher-1] Link base_scan had 0 children
[robot_state_publisher-1] Link wheel_left_link had 0 children
[robot_state_publisher-1] Link wheel_right_link had 0 children
[robot_state_publisher-1] [INFO] [1664735318.586357432] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1664735318.587990899] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1664735318.588248023] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-1] [INFO] [1664735318.588383746] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-1] [INFO] [1664735318.588491658] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-1] [INFO] [1664735318.588590663] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-1] [INFO] [1664735318.588693680] [robot_state_publisher]: got segment wheel_right_link
[turtlebot3_ros-3] [INFO] [1664735318.595027975] [turtlebot3_node]: Start Calibration of Gyro
[ld08_driver-2] /dev/ttyACM0    OpenCR Virtual ComPort in FS Mode
[ld08_driver-2] /dev/ttyUSB0    CP2102 USB to UART Bridge Controller
[ld08_driver-2] FOUND LiDAR_
[ld08_driver-2] LiDAR_ started successfully
[turtlebot3_ros-3] [INFO] [1664735323.596290544] [turtlebot3_node]: Calibration End
[turtlebot3_ros-3] [INFO] [1664735323.596640115] [turtlebot3_node]: Add Motors
[turtlebot3_ros-3] [INFO] [1664735323.598648695] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-3] [INFO] [1664735323.600147448] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-3] [INFO] [1664735323.616320352] [turtlebot3_node]: Succeeded to create battery state publisher
[turtlebot3_ros-3] [INFO] [1664735323.633280522] [turtlebot3_node]: Succeeded to create imu publisher
[turtlebot3_ros-3] [INFO] [1664735323.644526975] [turtlebot3_node]: Succeeded to create sensor state publisher
[turtlebot3_ros-3] [INFO] [1664735323.649147291] [turtlebot3_node]: Succeeded to create joint state publisher
[turtlebot3_ros-3] [INFO] [1664735323.649481551] [turtlebot3_node]: Add Devices
[turtlebot3_ros-3] [INFO] [1664735323.650206110] [turtlebot3_node]: Succeeded to create motor power server
[turtlebot3_ros-3] [INFO] [1664735323.663987948] [turtlebot3_node]: Succeeded to create reset server
[turtlebot3_ros-3] [INFO] [1664735323.669797079] [turtlebot3_node]: Succeeded to create sound server
[turtlebot3_ros-3] [INFO] [1664735323.678061703] [turtlebot3_node]: Run!
[turtlebot3_ros-3] [INFO] [1664735323.775078401] [diff_drive_controller]: Init Odometry
[turtlebot3_ros-3] [INFO] [1664735323.821200996] [diff_drive_controller]: Run!

```

9. Please describe the issue in detail.

When I power my turtlebot3 robot, the lidar is spinning by default. But when I run  ` ros2 launch turtlebot3_bringup robot.launch.py` command,  my lidar stops spinning and  not receiving any scan values in the ros2 topic echo /scan.
In the log file I see:
```
[ld08_driver-2] /dev/ttyUSB0    CP2102 USB to UART Bridge Controller
[ld08_driver-2] FOUND LiDAR_
[ld08_driver-2] LiDAR_ started successfully
```
but lidar not spinning and not send any data in topic /scan.

sunghowoo, 2024-11-15:
@DarekLin Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-11-15 by sunghowoo

muqtasid87, 2025-04-22:
@sunghowoo I'm using ROS2 Humble on my Turtlebot3 burger, and Ubuntu 22.04 on my pc and the server version apt for the raspi. im facing the exact same issue. has there been a solution?

GyuH13, 2025-04-23:
@muqtasid87 Hello
Could you please explain your situation?
As I understand it, the LiDAR doesn't spin when you launch the bringup, but there’s no error log. Is that correct?

muqtasid87, 2025-04-23:
yep I just receive the same message as OP described in point 8, which is just the standard bringup message.

Also initially I was trying with the LDS-02 and it wasn't working, but yesterday I tried with an LDS-01 and it worked. (yes i made sure my environment variables were correct for both cases)

So basically LDS-01 works as expected but LDS-02 doesn't. It just stops spinning.

GyuH13, 2025-04-24:
@muqtasid87 
Please clone the latest version of the lds08-drivers and launch the LiDAR node separately.
Then could you show me the log?